jks-prv/Beagle_SDR_GPS #1: Other users' frequencies should be clickable

elfchief, 2016-09-24:
In the panel where the current users are shown, the current frequency/mode shown for each user should be clickable, with a click setting your own RX's frequency/mode appropriately. This would make for a very nice "wonder what that person is listening to" kind of feature, with very little development effort.

jks-prv, 2016-09-24:
That's a great idea. I'll add it.

sm3sgp, 2016-09-28:
The new feature works well, thanks! 
Is it possible to copy the link in some way, to get it in a url format that is possible to copy?  The it would be easy to send a link to a friend to listen in. Example: http://fernebo.sk3w.se:8073/?f=3510cwnz11

jks-prv, 2016-09-28:
Hi.

Another good idea you have. Probably the best I can do is add a URL link icon, i.e. 

to the end of the line and add a URL link to that. The existing link has to be a javascript call and can't be a URL link at the same time. If it were only a URL link then when you click it a new connection to the SDR would be opened with that frequency instead of simply a re-tune. Definitely not what you want.

sm3sgp, 2016-09-28:
Or maybe a separate icon close to the frequency "input" window (right lower corner), to just copy the URL link with what your are listening to. I don't think it need to be  connected to what the others are listening to.

jks-prv, 2016-09-28:
Yes, I think you are correct. I think that should be easy.

elfchief, 2016-09-28:
You can actually have a button that works for both purposes -- You can use URLs with a # in them, with the various state information after the #.  You can then use javascript to both detect changes to the URL in the browser, and to set the browser's URL to match the current listening mode.  This means that you could cut & paste the browser's URL to give someone, or have a link in the listening stations list that is both clickable (which will change the URL but *not* reload the page, as long as only the bit after the # changes), and cut&pasteable.

[This](http://stackoverflow.com/a/7730202) stackoverflow comment talks about this kind of setup briefly. I couldn't find a good reference quickly, but a number of pages on 'single page web applications' talk about this technique.

jks-prv, 2016-10-02:
I added a link URL icon after the frequency entry box in the recent v1.8 release.

closed 2016-10-04 by jks-prv