jks-prv/Beagle_SDR_GPS #2: Allow listening when all four RXs in use

elfchief, 2016-09-25:
The number of unique streams leaving the KiwiSDR is limited by the processing power available to generate them, but sending the same stream out to multiple people should require almost no extra overhead. It would be very nice if, when all four RXs are in use, you were given the option to listen to an existing stream rather than being blocked completely.

jks-prv, 2016-10-02:
Interesting idea. Because the audio stream is compressed (ADPCM) there might be an issue with synchronizing a new connection tap to an existing stream. I think you just have to initialize the decoding state table correctly on the client. I can't remember exactly how that algorithm works.

G8JNJ, 2016-10-09:
Hi,

I also think this would be a very useful enhacement.

I know of a local radio club who are very keen to get a WEB SDR on line so that a number of members (>8) can all listen to the local net frequency at once. The current four user limitation is problematic for them, but this solution would be ideal.

Regards,

Martin - G8JNJ

jks-prv, 2023-04-16:
Added in v1.440 (March 2021)

closed 2023-04-16 by jks-prv