jks-prv/Beagle_SDR_GPS #3: 5kHz Step Up and Down

markwfahey, 2016-10-01:
Is it possible to add a 5kHz step buttons (5kHz step up & down in frequency) when in the AM modes.
Shortwave broadcast stations at 5kHz spacing and the current 1kHz and 10kHz steps while great retire additional key clicks if you wish to scan up or down the band by station channels.

jks-prv, 2016-10-02:
Should be fixed in the v1.8 release. Please let me know if this doesn't work in a useful way. I made the sideband modes step 5 kHz inside the SWBC bands also. I left the default at 1 kHz because that's generally what you want for the ham bands and probably for utilities. All this should be per-user configurable though..

closed 2016-10-02 by jks-prv

markwfahey, 2016-10-02:
I have tested this feature in v1.8. Thank you! Perfect - and such fast implementation - the project is the best, it really is a new revolutionary tool in the world of broadcast media (cultural/politcal/current events) montioring! :)