jks-prv/Beagle_SDR_GPS #4: Feature request: Listener IP-address privacy mode

gabofin, 2016-10-03:
I'd like to see possibility to enable privacy mode that hides listener IP-address from KiwiSDR WebUI. Currently default behavior is to display other listeners IP-address on WebUI. 

Listener IP-privacy mode should be able to turn on from admin panel. If IP-privacy mode is Enabled, KiwiSDR ui should

a) display salted hash of listener's IP-address instead of actual IP on WebUI. 

or 

2) Display text Anonymous text instead of listener's actual IP on WebUI.

jks-prv, 2016-10-03:
Yes, I agree. Showing people's IP and locations is leftover from the early days of development when I was curious where people were listening from. But you are correct: it's really a privacy issue where both Kiwi operator and user should have the choice.

How about also if you specify a user name of "anon" your IP is not listed. That way a user can decide if their IP is shown for each Kiwi they visit, independent of the privacy policy set in the admin panel.

jks-prv, 2016-10-03:
Okay, I looked at the code and refreshed my memory about how it currently works. The IP address is only displayed when the user leaves the "name or callsign" field blank. So the simple solution is just to display nothing when nothing has been entered. No configuration options needed. However IP addresses are still retained by the log files.

closed 2016-10-04 by jks-prv