jks-prv/Beagle_SDR_GPS #5: added frequency offset when click a carrier frequency at waterfall. (on the LSB, USB, CW, CWN mode)

icm7216, 2016-10-04:
# How to check this issue.

### 1: receive time signal. 

*   freq = 10000.00 kHz
*   mode = AM

### 2: a dial to show carrier frequency is below.

```
  /-------------------|-------------------\ 
  |    |    |    |    |    |    |    |    |
(mode = AM)       10.000 MHz
                      | <= carrier frequency
```

### 3: change a mode to LSB. and then click a carrier frequency at waterfall.

```
         /----------\ | 
  |    |    |    |    |    |    |    |    |
(mode = LSB)      10.000 MHz
                      | <= click
```

### 4: received frequency is moved to the upper. display frequency is 10001.52 kHz.

```
                /----------\ | 
  |    |    |    |    |    |    |    |    |
(mode = LSB)      10.000 MHz
                      |
```

Why added frequency offset? It should not move the received frequency. 
also USB, CW and CWN mode are similar problems.

I think that carrier frequency should be displayed as reception frequency.

jks-prv, 2016-10-04:
Hi. Yes, I expected this to be a controversial issue. In cw/cwn when I click on the keyed carrier in the waterfall I usually want the center of the passband to go to that frequency so I hear the beat note. For example, the keyed carrier is on the waterfall at 7021 kHz. When I click directly on it the center of the passband is set to 7021. But this means the actual receive frequency is 7021 - 500 Hz = 7020.5 kHz because the default cw passband is centered at 500 Hz (and usb mode). I display 7021 in the frequency box taking the 500 Hz offset into account. I think this strategy is probably what people want. When you click directly on a cw signal you should hear the offset tone (beat note), not the silence you would hear if you set the passband carrier on that frequency.

Now the question is what to do about usb/lsb. This is less clear. When I see a sideband signal in the waterfall I tend to click in the center of the width of the signal rather than trying to guess exactly where the lower edge is. Clicking on the center means that's where you want the passband center to go, just like the cw case. Because in sideband the carrier is suppressed you don't see a waterfall line where the carrier is. So you don't see something you can click on. Now it's true that most people set their transmitted frequency to the nearest 1 kHz. So that makes it easier. If you shift-click in the waterfall the passband is set to the nearest 1 kHz point rather than exactly where you're pointing (5 kHz in the broadcast bands).

But this behavior should probably be a user configuration option.

icm7216, 2016-10-05:
@jks-prv thanks. I could understand your explanation.

Usually, people want to click the center of the passband on the waterfall, when receive a sideband signal. Because suppressed sideband the carrier is not displayed in the waterfall. 

Yes, I think that in the sideband signal, this idea is a convenient way. 
This issue was resolved.

closed 2016-10-05 by icm7216

G8JNJ, 2016-10-09:
Hi,

I would prefer t obe able to set the carrier frequency and it remain fixed, so that when switching between AM, LSB & USB you don't have to retune.

Adding this as a user configuration option would be the best solution.

Regards,

Martin - G8JNJ