jks-prv/Beagle_SDR_GPS #7: Hide mode, freq, s-meter panel?

azwirko, 2016-10-06:
Is it possible to add a triangle, hide option to the right hand mode, freq select & s-meter panel to push it off into the drawer on the right hand side of the display, similar to what is done on the left for user list and intro/help lists?  It would be nice to see the entire display and be able to take screen shots without any blockage.

Thanks & 73

andyz - K1RA

jks-prv, 2016-10-06:
Yes I will do that.
Both WebSDR and OpenWebRX have better methods now for selecting which control panels to show. Perhaps we need something similar.

G8JNJ, 2016-10-09:
I'd also support this suggestion, as the HF end of the waterfall is the most useful as stations only appear when conditions are good. It's currently a nuisance that the panel obscures this part of the screen.

Could you also place the extensions panel above the control panel and make it slide away too.

Regards,

Martin - G8JNJ

G8JNJ, 2016-11-08:
Hi John,

I've had some further thoughts about the placement of the control panel. 

The main priority for me is to make it slide away so that it's not obscuring the HF end of the waterfall, as I find this to be really annoying becuase the HF end is where the weak short duration signals tend to be present, as opposed to the Broadcast stations which are always present at a high level on the left hand side of the screen

However I think it would be better from a cosmetic and operational perspective, if the control panel was placed along the whole length of the bottom of the waterfall, so that the panel was not as tall as at present but wider. This would obscure much less of the waterfall when the panel was open.

It would also be good if the control panel could drop down but have a wide strip still showing along the bottom of the waterfall that could be used to pop it back up (or down) a bit like the windows task bar. This would reduce the amount of mouse movement required to bring the panel back up for the most common operations.

Regards,

Martin

epninety, 2016-12-11:
In a similar vein, if the level callout for the marker in the spectrum display could be below the pointer, it wouldn't get hidden by the browser address bar or disappear off the top of the screen on strong signals.
I know I can rescale, but it's nice to keep as big a spectrum display as possible when trying to look at levels accurately (I'm comparing my various active antenna builds and investigating spurious signals).

G8JNJ, 2017-04-16:
Hi John,

Thanks for including this request in V1.71 :+1: 

Works great and really tidies up the waterfall when on full width.

Regards,

Martin - G8JNJ

closed 2017-04-16 by jks-prv