jks-prv/Beagle_SDR_GPS #9: Feature request: IPv6 support to kiwid

gabofin, 2016-10-08:
kiwid listens only on IPv4 socket. It does not bind to IPv6 socket. It would be nice to have IPv6 support in kiwid.

Until kiwid has native support there is workaround to get IPv6 work. Install socat to beaglebone and make IPv6 -> IPv4 proxy.

socat TCP6-LISTEN:8073,reuseaddr,fork,bind=[2001:DB8::BEA6:73] TCP4:127.0.0.1:8073 

then KiwiSDR seems to work fine with IPv6.

jks-prv, 2016-10-08:
Hi. So we just had this patch which you may not have seen: https://github.com/jks-prv/Beagle_SDR_GPS/pull/8
I'm not that familiar with all the IPv6 issues. Does this patch have the same effect as your proxy solution? The patch will be included in the v1.11 release.

gabofin, 2016-10-09:
> Does this patch have the same effect as your proxy solution? 

Yes. It looks to me that I do not need proxy solution after v1.11 release anymore.

gabofin, 2016-10-11:
I run version 1.11 on my site http://sdr.vy.fi now. Both protocols IPv4 and IPv6 seems to work fine. No need to socat proxy anymore. Thank you!

closed 2016-11-30 by jks-prv