jks-prv/Beagle_SDR_GPS #11: Tags and bands

G8JNJ, 2016-10-09:
Is it possible for SDR admins to customise / edit the frequency tags and bands ?

Is there any documentation ?

Regards,

Martin - G8JNJ

G8JNJ, 2016-10-09:
OK I found how to edit the tags Shift click on tag when logged in as admin.

Would it be possible to edit the band information in the same way.

Regards,

Martin - G8JNJ

G8JNJ, 2016-10-23:
Hi John,

I've been gradually adding more tags to my SDR and it's now starting to look a bit confusing.

I know that you intend to revisit the tagging method, but I wondered if it would be possible to make individual tags stand out when the cursor is moved across (within the passband of whatever mode has been included in the tag) a tagged frequency on the waterfall.

This would allow folks to see if that the user of the frequency had been identified and it would make it easier for admins to spot new frequencies that still require tagging.

Regards,

Martin - G8JNJ

jks-prv, 2016-10-23:
That's an interesting idea. SImilar to how tags highlight yellow when you mouse over them now.

ka7u, 2016-10-28:
What is this pop up tag, that seems to randomly show up when tuning in 100Hz steps?
[image]

jks-prv, 2016-10-28:
Hi Ron. What browser & OS is that? Does this popup happen when you clicked the 100 Hz (innermost) +/- buttons? And it persists until you close it with the 'X' at the upper right of the popup? When you say 'randomly' do you mean it doesn't show up 100% of the time you click the 100 Hz step buttons?

Looks like another case of a browser doing something different and unexpected from all the others.

ka7u, 2016-10-28:
John, It happens rarely, and the window disappears on the next tuning step or mouse click. It will also disappear if I click the "X". I'll watch other systems and see if it is unique to this openSUSE system. It is not an important operating concern.

Version 54.0.2840.71 (64-bit) Google Chrome
**$cat /etc/os-release**
NAME=openSUSE
VERSION="13.2 (Harlequin)"
VERSION_ID="13.2"
PRETTY_NAME="openSUSE 13.2 (Harlequin) (x86_64)"
ID=opensuse
ANSI_COLOR="0;32"
CPE_NAME="cpe:/o:opensuse:opensuse:13.2"
BUG_REPORT_URL="https://bugs.opensuse.org"
HOME_URL="https://opensuse.org/"
ID_LIKE="suse"
 **$kde4-config -v**
Qt: 4.8.6
KDE Development Platform: 4.14.9
kde4-config: 1.0

jks-prv, 2016-10-28:
Okay, thanks. Some Googling around shows the message "no definition found" also comes from iOS / OSX when a dictionary lookup fails. On iOS you start such a lookup by selecting something (like double tapping on the frequency box), a menu comes up, and one of the menu entries is "define". The define will obviously fail for just a number, 7167.5 in your case. OSX has a similar entry in the menu that comes up when you right-click with the frequency box selected.

Now the question is why Chrome on SUSE is doing this spontaneously. My Chrome on OSX will bring up a menu if you control-click with cursor over the 100 Hz step buttons. But that doesn't end up with a "define" menu action happening. So this is not clear..

ka7u, 2016-10-29:
John, I've tried 3 other computers running Chrome, Ubuntu, and Windows. These machines do not produce the window. So I conclude it is a specific browser issue with the openSUSE computer. Case closed, I think.

jks-prv, 2023-04-16:
Fixed in v1.507 (April 2022)

closed 2023-04-16 by jks-prv