jks-prv/Beagle_SDR_GPS #12: Saving and recalling config from admin menu

G8JNJ, 2016-10-09:
Is it possible to add a "save to" and "load from" SD card option within the SDR admin pages ? 

This would save having to modify files on the SD card then reload from scratch and await updates. 

Regards,

Martin - G8JNJ

jks-prv, 2016-10-09:
Could you explain further? The SD card is intended to be a backup of the factory delivered configuration (v1.4 I think) and not overwritten.

If you are wanting to make a backup of your Beagle to a new SD card a very easy way is to use the Kiwi manufacturing interface at: kiwisdr.local:8073/mfg Use the cyan-colored button labeled "click to write micro-SD card". The resulting SD card will be a "flasher" just like the backup SD card supplied with the Kiwi. In fact, this manufacturing interface is exactly how Seeed makes those backup cards at their factory.

G8JNJ, 2016-10-09:
Hi John,

OK this is exactly what I wish to do.

Ignore my request and close.

Thanks for the information.

Martin - G8JNJ

jks-prv, 2016-10-10:
Hi Martin,

I added "backing up Beagle to SD card" to the title. I think this should be a feature available directly on the admin page and not only buried on the manufacturing interface. Especially if admins invest a lot of time customizing their Kiwis they are going to want a simple way to backup, and a simple way to restore.

G8JNJ, 2016-10-11:
Hi John,

Excellent news - I think this will be very useful and save a lot of time for admins.

Regards,

Martin - G8JNJ

G8JNJ, 2016-10-19:
Hi John,

I tried backing up to an SD card using the manufacturing interface.

It all ran well for about 10mins untill it got to 95% then it just sat there with the time counter running.

I stopped the process after about 1 hour as it was obvious that it wasn't going to finish.

Does the SD card need to be specially formatted before it can be used ?

Regards,

Martin - G8JNJ

jks-prv, 2016-10-19:
No format of the card required. The script does everything. Give it another try. It occasionally gets stuck when I use it. If it happens again I'd be curious to know what the last messages are in the log pane at the bottom of the window.

mmagin, 2016-10-19:
I would definitely make sure that you are using a name-brand
(e.g. Samsung, Sandisk, Kingston, Lexar, etc.) and non-counterfeit micro
SD card.  There are a lot of fake SD cards out there in the wild and
they lie about their capacity and/or have other problems.

Further (and maybe John can comment on whether this is an issue for the
Beaglebone as configured in KiwiSDR), I know that some devices have
compatability issues with the UHS speed class cards (they're supposed
to downgrade smoothly, but a number of Android users had problems in the
past.)  https://en.wikipedia.org/wiki/Secure_Digital#Speed_class_rating
In that case, using one of the regular class 2 through class 10 cards
is probably less problematic.

jks-prv, 2016-10-19:
Mike is absolutely correct. The quintessential (ha! I get to use that word in a sentence) article on the problems with counterfeit SD cards is this one I found long ago: https://www.bunniestudios.com/blog/?page_id=1022
It's a fascinating read.

G8JNJ, 2016-10-19:
OK - good points I'll try another couple of cards.

Regards,

Martin - G8JNJ

G8JNJ, 2016-10-21:
Hi John,

Still no luck even with branded cards.

[image]

Regards,

Martin - G8JNJ

jks-prv, 2016-10-21:
There is something wrong here. That light grey panel at the bottom should be much larger and show at least the starting messages from the copy. Does the small vertical scroll bar scroll down at all?

G8JNJ, 2016-10-21:
Hi John,

No it doesn't scroll.

Regards,

Martin - G8JNJ

jks-prv, 2016-10-26:
I'm going to reopen this because after some email exchanges with Martin it really is a bug (an assertion fails in mongoose) and is a feature I'd also like to see working.

G8JNJ, 2017-04-20:
Hi John,

I'm not sure WRT the current status of this bug.

However I tried a backup today using the Admin interface and it still sticks at 95%.

Here's a grab of the gray log screen.

Regards,

Martin

ADMIN: received microSD_write
-----------------------------
KiwiSDR: copy eMMC image to micro-SD card
Version: [1.20151007: gpt partitions with raw boot...]
-----------------------------
include /root/Beagle_SDR_GPS/tools/kiwiSDR-SOC.sh
Unmounting Partitions
----------------------------
copying: [/dev/mmcblk0] -> [/dev/mmcblk1]
lsblk:
NAME MAJ:MIN RM SIZE RO TYPE MOUNTPOINT
mmcblk0boot0 179:8 0 2M 1 disk 
mmcblk0boot1 179:16 0 2M 1 disk 
mmcblk0 179:0 0 3.7G 0 disk 
`-mmcblk0p1 179:1 0 3.7G 0 part/
mmcblk1 179:24 0 3.7G 0 disk 
`-mmcblk1p1 179:25 0 800M 0 part 
-----------------------------
df -h | grep rootfs:
-----------------------------
actual rootfs size = 694M
preset micro-SD partition size = 800M
for backup setting micro-SD patition size to actual rootfs size 800M + 100M
update-initramfs: Generating /boot/initrd.img-4.4.9-ti-r25
Erasing: /dev/mmcblk1
Erasing: /dev/mmcblk1 complete
-----------------------------
Writing bootloader to [/dev/mmcblk1]
dd if=/opt/backup/uboot/MLO of=/dev/mmcblk1 count=1 seek=1 conv=notrunc bs=128k
-----------------------------
-----------------------------
dd if=/opt/backupuboot/u-boot.img of=/dev/mmcblk1 count=2 seek=1 conv=notrunc bs=384k
-----------------------------
-----------------------------
Formatting: /dev/mmcblk1
sfdisk: [sfdisk from util-linux 2.25.2]
sfdisk: [sfdisk --force --Linux --in-order --unit M /dev/mmcbk1]
sfdisk: [1,794,L,*]

Disk /dev/mmcblk1: 121600 cylinders, 4 heads, 16 sectors/track
Old situation:
New situation:
Units: 1MiB = 1024*1024 bytes, blocks of 1024 bytes, counting from 0

Device Boot Start End MiB #blocks Id System
/dev/mmcbk1p1 * 1 794 794 813056 83 Linux
/dev/mmcblk1p2 0 - 0 0 0 Empty
/dev/mmcblk1p3 0 - 0 0 0 Empty
/dev/mmcblk1p4 0 - 0 0 0 Empty
Successfully wrot the new partition table

Re-reading the partition table ...

mkfs.ext4 /dev/mmcblk1p1 -L KIWISDR
-----------------------------
Discarding device blocks: 4096/203264 failed - Input/output error
Creating filesystem wth 203264 4k blocks and 50848 inodes
Filesystem UUID: fe3ff455-4e85-4790-a0f4-a65eef5a2f0a
Superblock backups stored on blocks: 
32768, 98304, 163840

Allocating group tables: 0/7 done 
Writing inode tables: 0/7 done 
Creating journal (4096 blocks): done
Writing superblocks and filesystem accounting information: 0/7

jks-prv, 2017-04-20:
Yes, it will "stick" at 95% until finished under certain circumstances. I can't know how long the copy process will take because it varies with how full your Beagle filesystems is (although they tend to be similar unless the log file has grown huge) and how fast the SD card you're using is.

I measured a typical time of 4 minutes when using a class 10 SD card. So that defines what 100% is. If it takes longer than that I make the progress bar stick at 95% (rather than 100%) and jump to 100% at the end. If it finishes before that the progress bar jumps to 100% when done. The copy process itself is a script that does a bunch of things. So it's not just one Linux command that can tell me how far along it is. It the script gets hung for some reason, then yes, it will stay at 95% forever.

G8JNJ, 2017-04-21:
Hi John,

OK understood. 

I tend to wait with it stuck on 95% for about 10 mins before giving up.

I've now tried three different makes of cards but they have all been used sometime before.

I'll buy another new one and see if that works.

Regards,

Martin - G8JNJ

jks-prv, 2017-04-21:
Try again with the v1.76 release. While redoing the backup tab UI I had it hang at the point your log shows. It was because the Kiwi server had crashed. I thought this was to a new bug I had introduced, but it is possible this bug was present in v1.75 as well and this is what you are seeing.

G8JNJ, 2017-04-22:
Hi John,

Tried again with V1.77 sticks at 95% even after 20 mins.

Haven't got a new card yet.

Anyone else seen this, or maybe no one backs up their KiWi ?

Regards,

Martin - G8JNJ

jks-prv, 2017-04-22:
Well I don't know what to say. I just tried it with three 8GB SD cards. Two different manufacturers. And they all worked fine.

Is the last message you see in the little log window as in your message above? "Writing superblocks and filesystem accounting information: 0/7" You may have to scroll it to the bottom.

jks-prv, 2017-04-22:
Others may not have been backing up because until the last few releases there was no backup tab on the admin page. You had to know to bring up the /mfg page which I don't think most people know about.

jks-prv, 2017-04-22:
After it has hung up, if you switch tabs to "status" (first one) does the "uptime" continue to increment? (it only does this every ten seconds). How about if you refresh the page to make a new admin connection? Do that still work? I'm wondering if the Kiwi server has crashed and that's why the backup appears hung. The spinning icon happens in the browser and is not an indication that the server is actually responding.

G8JNJ, 2017-04-23:
Hi John,

The Uptime is incrementing and if I refresh I get a connection.

I tried two cards (4G and 2G) using the GUI interface. Both got stuck at 95% 

'Writing superblocks and filesystem accounting information: 0/7���'

I then tried both cards using Putty and the command line

./tools/kiwiSDR-make-microSD-flasher-from-eMMC.sh

One (4G) failed at the same point 'Writing superblocks' but the other (2G) proceeded beyond this point and completed the backup.

So one card could be bad but the other should have worked.

Regards,

Martin - G8JNJ

closed 2017-06-03 by jks-prv