jks-prv/Beagle_SDR_GPS #13: Adding frequency offset for down-convertors

G8JNJ, 2016-10-09:
Hi,

Would it be possible to add a frequency offset so that the SDR could be used with block down convertors to provide coverage of higher frequency bands ?

Regards,

Martin - G8JNJ

jks-prv, 2016-10-09:
Yes, this request was also made by Ben, PE2BZ, over on the Forum. He also asked for the current range extension option to 32 MHz. He's now downcoverting 430-437 to 25-32 MHz on his Kiwi.
http://valentfx.com/vanilla/discussion/425/placing-my-own-labels-on-my-kiwisdr

Thanks for getting this feature request onto this list.

G8JNJ, 2016-10-10:
Would it be possible to also add a feature that would allow the tuning range to be split into separate frequency bands each with a different frequency off-set. This would allow multiple down-convertors with different input and output frequencies to be used simultaneously, or a mix of some HF and VHF / UHF bands to be implemented.

e.g. 50-52MHz -> 5-7MHz  / 70-71MHz -> 10-11MHz / 144-148MHz -> 14-18MHz / 433-440MHz -> 23-30MHz

This would be used in conjunction with issue #14 

Regards,

Martin - G8JNJ

BjarneBachmann, 2017-02-12:
Hi
Remember that the mixing frequency could be placed so that a higher input frequency would give lower frequency to the KiwiSDR ! Maybe a negative offset could be used.

jks-prv, 2019-01-18:
You can configure a single-range, positive-offset down conversion parameter on the config tab of the admin page. See the "frequency scale offset" field.

jks-prv, 2023-04-16:
"Downconverter high-side injection" checkbox was added to admin page, config tab, in v1.550 (Aug 2022).
Split bands is still an open idea.