jks-prv/Beagle_SDR_GPS #14: Restricting frequency ranges

G8JNJ, 2016-10-09:
Hi,

Would it be possible to add the option to restrict the tuning range to be within defined upper and lower limits ?

It would be useful if other blocked frequencies could also be defined by pairs of upper and lower limits.

This would help with regulatory requirements in some countries.

Regards,

Martin - G8JNJ

jks-prv, 2016-10-09:
I've often wondered about this. I assumed these days it was only an issue at V/UHF.

G8JNJ, 2016-10-10:
It is mainly for VHF / UHF use but there are other instances where it may be useful.

I also act as Admin for another WEB SDR and we have had an instance of someone using the SDR to listen to a local net and then use it to remotely listen whilst jamming one or more net users. 

Adding a feature like this would allow specific frequencies to be locked out if required.

The other use would be to allow multiple down-convertors with different output frequencies to be used simultaneously. The unwanted frequency bands in-between the down converted bands could then be locked out.

Regards,

Martin - G8JNJ

G8JNJ, 2016-10-16:
I've now got a lot of Dutch users who are using my SDR to listen to 6MHz pirate music stations. I don't have a problem with what they are listening to, but they tie up the SDR for hours on end and effectively lock out other users. 

The ability to restrict a small frequency range around a specific center frequency would be useful in such an instance.

Regards,

Martin - G8JNJ

jks-prv, 2016-10-16:
Yeah, they do this on Twente WebSDR too. Good thing that SDR can handle 400 simultaneous connections, lol.

G8JNJ, 2016-10-18:
I have been thinking further about this suggestion.

Would it be possible to add a time limit to the frequency lockout ?

This would be a lot less confrontational than entirely preventing listening.

This would allow users to initially listen to a locked out frequency range for a defined time period, before the SDR locked them out from further listening on that specific frequency range.

The time could be defined in seconds. 0 seconds  = no listening.

Regards,

Martin - G8JNJ

jks-prv, 2021-02-27:
So we more or less have this now. Create a DX label with type = "masked".
See also: http://forum.kiwisdr.com/index.php?p=/discussion/comment/10209

closed 2021-02-27 by jks-prv