jks-prv/Beagle_SDR_GPS #15: Loading new photo

G8JNJ, 2016-10-09:
In which directory should the title photo reside ?

I've tried uploading my own picture which is the same size as the original pcb.jpg but it won't load.

As a test I have tried replacing root/Beagle_SDR_GPS/web/kiwi/pcb.jpg with a copy called pcb2.jpg and setting that as the default through the admin interface but the file can't be found.

Does a replacement photo need to go in another folder ?

Regards,

Martin - G8JNJ

jks-prv, 2016-10-09:
This is described in the "Webpage" tab section of the quick start guide: http://kiwisdr.com/quickstart/index.html#id-config
Basically, the photo file goes in /root/kiwi.config because /root/Beagle_SDR_GPS is subject to being completely overwritten by an update.

G8JNJ, 2016-10-09:
OK I have loaded a new photo called coast2.jpg into /root/kiwi.config/coast2.jpg

But I still get a 404 file not found

G8JNJ, 2016-10-09:
Ooops - forgot to change the photo file location to kiwi.config in the web admin page.

All OK now - you can close this one too - user error :-(

Regards,

Martin - G8JNJ

jks-prv, 2016-10-09:
For the wish list: add a "file picker" (or whatever they're called) for the photo field so you can download the file to the proper place via the admin UI.

G8JNJ, 2016-10-09:
Good idea :-)

Regards,

Martin - G8JNJ

closed 2016-11-30 by jks-prv