jks-prv/Beagle_SDR_GPS #16: Set number of public recievers

G8JNJ, 2016-10-10:
This is applicable if the suggestion of being able to permit multiple users to listen to existing receivers is implemented.

Would it be possible to somehow set password protection or some sort of priority use per receiver in the admin menu ?

Would it also be possible to set a default frequency and mode to individual password protected receivers ?

This would allow authorised users priority over other casual users on one or more receivers if required.

My thinking is that if a radio club operates a KiWi SDR they may wish to leave one receiver permanently set on their local net frequency.

Regards,

Martin - G8JNJ

jks-prv, 2016-10-10:
It has been suggest that because rx channels are such a precious resource there should be a property "matrix" that allow independent settings for each channel. Things like if a password is required, Internet uplink bandwidth limitations, reserved for distributed experiments, and as you suggest, set to a dedicated frequency and/or allows multiple passive connections.

G8JNJ, 2016-12-22:
Hi John,

I suspect (but cannot 100% confirm) that the 'Number of channels not requiring a password even if assword set' function is not working in V1.33.

This evening I have four users on my WEB SDR, and I can't log into the WEB SDR locally even though I have three public receivers set and one password protected. 

I don't even get the password screen presented to me in order to have the option be able to log in.

Regards,

Martin