jks-prv/Beagle_SDR_GPS #17: Admin page password not letting me on. I'm on local network.

chuckfletcherIV, 2016-10-10:
Docs say it doesn't require a password if you are on the local network, but mine is. Any ideas.

Chuck

jks-prv, 2016-10-10:
So let me understand: You can't get in to the admin page at all? Or you set a password, can get in using it, but expect to get in without using it because the "auto-login from local net even if password set" option is set? Thanks.

chuckfletcherIV, 2016-10-10:
Can't get into the Admin section via browser at all, but I'm on the same network. 

Kiwi IP: 192.168.0.144. 
Netmask: :255.255.255.0

Mac with Browser: 192.168.0.109.

Is this something I could fix from the command line?

jks-prv, 2016-10-10:
Hmm, nothing wrong with that. Yes, there's some stuff we can look at via shell commands on the Beagle. Please email me at support@kiwisdr.com
We can post the solution here for others to see. Thanks!

jks-prv, 2016-10-31:
We think this has been fixed with the addition of IPv6 local network detection code in the v1.18 release. If anyone continues to have trouble with this please be sure to let us know.

closed 2016-10-31 by jks-prv