jks-prv/Beagle_SDR_GPS #18: I'd like to be able to send spots to PSK reporter right within KiwiSDR.

chuckfletcherIV, 2016-10-10:
https://www.pskreporter.info/pskmap.html

If this isn't possible, I'm guessing I might be able to get it send the audio to something like FL-DIGI, but might need to send some rig control signals to make FL-digi be in sync so far as freq. 

Thanks for any ideas on this,

Chuck

jks-prv, 2016-10-10:
This would make for a good extension and I'll add it to the list. I don't know much about PSK31. We would need (open) source code to a decoder to write an extension. A quick Google shows Moe Wheatly, AE4JY, has one. We use parts of his CuteSDR in KiwiSDR already. http://www.moetronix.com/ae4jy/pskcoredll.htm

chuckfletcherIV, 2016-10-10:
Cool. It might make sense to think about what happens internally and externally with KiwiSDR. I'm sure there are common things you should decode right in the web browser interface, but also it would be great if there was a common way to send data to CuteSDR or FL-digi and just stream out the audio (maybe with a virtual audio port or JACK on Mac and LInux). It's important that frequency data is sent along with the audio so that the SDR apps know your intended frequency range and etc. For example -  when I change frequencies on my HF radio (by hitting a button or tweaking a dial) it updates the SDR app info using either Hamlib, RigCat or XML-RPC. This is a bit simpler on a recieve only SDR like KiwiSDR because you don't need to control the radio for TX.

jks-prv, 2016-10-10:
Yes, there are already open requests for IQ output, rig control protocols (NetSDR et al), documentation of the full network API so arbitrary client code can be used instead of the javascript/HTML5 browser interface, etc. Someone even wants an extension with a bunch of buttons that can control an antenna switching box via USB. Lots that needs to be here.