jks-prv/Beagle_SDR_GPS #19: Suggestion for possible extension - Morse / CW decoder / Skimmer

G8JNJ, 2016-10-12:
Hi,

I wondered if it would be possible to include a Morse / CW decoder / Skimmer as an extension ?

There's still a lot of it about, and it would be useful for identifying NDB's and suchlike.

Regards,

Martin - G8JNJ

jks-prv, 2016-10-12:
Yes, would be nice. The existing Skimmers are all closed-source right? I'm pretty sure Alex's, VE3NEA, is closed. Can't blame him at all. It's a pretty neat hack!

G8JNJ, 2016-10-12:
Hi John,

Yes Alex's skimmer is closed source and it does work very well. I've yet to find anything else that is as good.

I was thinking of perhaps something much simpler that could be used to decode just one CW transmission ? 
 
Regards,

Martin - G8JNJ

rikvanriel, 2016-10-24:
Whatever we start with does not have to be as good as that proprietary skimmer.

Starting with something simple, and enhancing it over time, is exactly the kind of thing open source development is good at. Are there any open source cw skimmers out there that could be integrated?

ka7u, 2016-11-03:
This image shows Fldigi monitoring the CW audio from the KiwiSDR web interface on an openSUSE OS. Pulse Audio is directing the audio connections and so Fldigi is the "recorder". The same could be done using virtual audio cables in Windows. If there were virtual serial ports such as com0com.exe provides in windows or SoCat provides in Linux, moving frequency and mode data to and from 3rd party applications such as Fldigi, then the KiwiSDR could be tuned (frequency and mode) from the 3rd party application and the 3rd party application could be tuned (frequency and mode) from the KiwiSDR web-browser waterfall input. This would allow for a great diversity of commonly available 3rd party applications to compliment KiwiSDR. That would be any instance of any KiwiSDR displayed on a web-browser using about any OS.
[image]