jks-prv/Beagle_SDR_GPS #21: Provide higher priority to local user

G8JNJ, 2016-10-12:
Hi,

My KiWi is being fully utilised this evening and I can't easily log in to test something.

Rather than dropping exiting users by restarting the SDR, would it be possible to give priority to users on the local network above those logging in via the internet ?

Regards,

Martin - G8JNJ

jks-prv, 2016-10-12:
Thinking out loud here: Maybe in this new "matrix" of per-channel options on the admin page there should be a "kick" button that allows you to close the connection of a particular channel (rather than closing all of them by restarting). An easy way to kick off a user with a large connect time or make room for yourself to connect.

G8JNJ, 2016-10-12:
It would be good to be able to kick folks off individually and very useful if there was a 'repeat offender' who you wished to dissuade from hogging the SDR.

Regards,

Martin - G8JNJ

dish-eme-sdr, 2017-10-17:
I haven't tested this extensively with kiwisdr, but I did use it extensively when running openwebrx on a raspberry pi-3 with four RTL-SDR rx to selectively kick individual users who insisted on hogging the system.  It is IP based and would of course not prevent the same user from reconnecting from a different IP, BUT it does permit the sysop to selectively kick a particular user thus freeing up a slot for use without kicking everyone else :-^)...

This method assumes the following::

1 = You have access to the command prompt: root@kiwisdr:~#
2 = You know the IP number of the offending user and want to kick that user while NOT disturbing the connection to other users.
3 = For this example the IP number of the offending user is assumed to be: 111.222.333.444

Enter the following at the command prompt root@kiwisdr:~#
iptables -A OUTPUT -s 111.222.333.444 -j DROP

To see a list of IP numbers that are presently blocked, type the following:
iptables -L

To remove a blocked IP from the list (permit that user to log back in):

iptables -D OUTPUT 1

"1" assumes the first entry in the IPTABLE.  (You can have more than one blocked IP address at a time and may want to clear IP 2 while leaving IP 1 still blocked, etc.)

It is possible to block connections by INPUT or OUTPUT.  Using OUTPUT does NOT block INPUT connections from the blocked IP number.  This permits the owner of the system to monitor connection attempts from the blocked user but the blocked user can never connect as they never see any output from the system.  Blocking by INPUT makes the offending user go away completely which also means there is no way for the sysop to tell if that IP is still trying to connect.

As with all commands, you can get help by typing:

iptables --h

NOTE: If you decide to try this on a raspeberry pi, you will most likely need to use sudo:

sudo iptables -L

Rebooting the raspberry pi cleared IPTABLE (no blocked users).  I assume kiwisdr would be the same, but have not tested this.  There are ways of restoring IPTABLE when rebooting but that is probably more than what is needed for what is being discussed here....