jks-prv/Beagle_SDR_GPS #22: Waterfall "logging"?

peejii73, 2016-10-13:
Hi,

first of all my most sincere thanks for this great project! In a nutshell: This little beast is absolutely super.

Say, what do you think of a feature similar to Twente's SDR: http://websdr.ewi.utwente.nl:8901/fullday/

That would be great in for instance trying to trace the nowadays far too frequent chinese SMPS QRM and similar, not mentioning the educational purposes: that would be nice to show people at local ham radio course, when the topic is radio propagation!

One proposal more over the Twente's display: how about adding whole picture width dotted lines or similar to show full hours? Now they disappear when this huge picture is scrolled to the right..

Thanks + 73,
-petri OH1BDF

jks-prv, 2016-10-13:
That's a good idea. I'll have to re-read P.T. description of how he does this, like whether any averaging or peak detection is used instead of merely sampling at a low rate.

G8JNJ, 2016-10-14:
Hi,

This is an interesting idea.

One of my KiWi SDR users has recently asked if it's possible to be able to adjust the waterfall speed to slow it down. I'd also like to be able to speed it up a bit so that I can visually ID NDB beacons sending slow morse letter ID's

Regards,

Martin - G8JNJ

jks-prv, 2016-10-14:
Waterfall speed control is now on the master bug/wish list: kiwisdr.com/bugs