jks-prv/Beagle_SDR_GPS #23: Problem with AM reception

G8JNJ, 2016-10-15:
This issue was flagged up by one of my SDR users.

When using AM to listen to a plain carrier signal, if you offset the tuning by say 1KHz you can hear a 1KHz beat tone. This is problematic when listening to broadcast stations or NDB's

For example tune to a plain carrier using USB set it to produce an audio tone of say 500Hz. Then switch to AM and you can hear the same frequency tone. It's almost as if the AM detector is actually a DSB detector. 

This should not happen when using AM. No tone should be heard.

I have tried the same experiment on an Openwebrx SDR and the problem doesn't seem to be present.

Regards,

Martin - G8JNJ

jks-prv, 2016-10-15:
I thought this problem had gone away some time ago, but I think I understand it more fully now. It is caused when the passband straddles the 0 Hz (carrier) point, which it would for a typical AM-mode filter. You can see the problem really clearly as follows: Tune to a quiet part of the band in USB mode. Bring up the IQ display and adjust the gain until you get a circular pattern of noise. Note that the pattern is symmetric WRT the crosshairs of the display. Now switch to AMN. Note that the pattern shifts to the upper left. Now drag one side of the AMN passband until it just crosses over the carrier point. The pattern will become symmetrical again.

So there is a slight IQ imbalance. I think there must be a problem with how I'm using the CuteSDR FIR filter to generate the adjustable passband.

jks-prv, 2016-10-15:
The other clue about this is that it is most noticeable for signals with carriers weaker than about -80 dBm. As you would find for most typical NDBs. You don't hear the carrier bleed-through on a strong AM BCB station where the carrier IQ values would be much larger that the IQ offset errors.

jks-prv, 2016-10-16:
Wow. Short story: will be fixed in v1.13
Longer story: this is not a simple bug with CuteSDR FIR I was expecting. In fact the FIR **fixes** the bug when the passband is set _not_ to pass the 0 Hz point of the audio spectrum, as we have observed. This is a _hardware_ issue. It's a small DC IQ offset coming from the ADC, ADC preamp or input circuit.

It seems to be very similar across all the Kiwis I've tested. This means it can be cured by simply adding a correction factor to the IQ samples. But I've also added a mode in the IQ display extension (if you're an admin) to measure the actual offset and add it to the overall correction factor. You have to disconnect the antenna so the measurement will be accurate. The default correction factor seems to be good enough to solve the carrier bleed-through problem, so most Kiwi owners won't have to do anything. When I get more time I need to understand the hardware basis of the problem.

G8JNJ, 2016-10-18:
Great thanks for doing this.

Just checked the AM reception.

I applied the correction and it's a lot better now :+1: 

Regards,

Martin - G8JNJ

jks-prv, 2016-10-18:
Okay, good. I'm hoping the default correction will be sufficient for most Kiwi boards. I still don't understand why there would be similar, small DC offset common to all boards unless there is a hardware design flaw. More investigation needed when I have time.

You can also fine-tune the correction using the IQ display. Disconnect the antenna and tune to a frequency that has a nice flat, low noise floor free of any spurs or other artifacts on the waterfall. Set to AM mode. Bring up the IQ display. Adjust the gain slider until you get a circle of noise points filling 2/3 of the display. It should be centered on the crosshairs. When everything looks okay hit the "clear" button and let the Kiwi sit for a minute or two. The CMA numbers under the IQ display will show the cumulative moving average of all the IQ samples during the measurement period. Because the default correction is 5e-2 the measured numbers should be less and will be added to the 5e-2 as a fine-tuning when you hit the "IQ val" button. The new value is stored in the configuration file.

G8JNJ, 2016-10-18:
Hi John,

Superb :+1:  I'll give that a go and see if it gets rid of the tiny bit of residual tone I can just about still hear.

Regards,

Martin - G8JNJ

closed 2016-10-26 by jks-prv