jks-prv/Beagle_SDR_GPS #25: Peak signal S-Meter display and possibly peak hold

G8JNJ, 2016-10-17:
Hi,

I frequently use the SDR's to compare signal strengths of stations participating in nets or to compare antennas. This is tricky due to signal fading and other variations in propagation.

Would it be possible to add a variable decay time, or a peak hold button, to the peak reading on the S-Meter so that it could be used to register the maximum received level over a longer period of time such as 1 minute or more ? 

I also wondered if it would be possible to add some sort of peak hold trace, possibly dots or a translucent line to the spectrum display ?

Regards,

Martin - G8JNJ

jks-prv, 2016-10-17:
I was going to write an extension that worked the same as Twente WebSDR's "S-meter plot" display. A nice signal strength "strip chart" with variable speed, time stamps and auto-scaling.

G8JNJ, 2016-10-17:
Hi John,

That would be good, but could the amplitude scale just be defined by the existing max / min sliders ?

Having auto gain is a bit of a pain as the scales bob up and down depending upon which station is operating. I'd prefer to be able to fix the scale limits.

Also could the graph have some sort of rolling average line in addition to a peak reading display.

Regards,

Martin - G8JNJ

jks-prv, 2016-10-17:
Sure, all good ideas, and all possible.

G8JNJ, 2016-11-21:
Hi John,

The Signal Strength plotting extension is very good, I have used it a few times already.

Maybe the scale calibration could be changed in the extensions S-meter graph configuration screen ?

Regards,

Martin

jks-prv, 2016-11-21:
Hi. That same update added separate S-meter and waterfall cal adjustments to the admin > config panel. I'm still getting the bugs out. The S-meter cal takes effect immediately, even on existing connections. The WF cal requires a connection reload (trying to fix that). The S-meter graph is obviously based on S-meter data, so updates immediately when the cal changes. These are config settings that apply to the Kiwi and effect every connection.

Are you asking for an S-meter graph extension cal that would be private to each user/connection? Certainly possible..

G8JNJ, 2016-11-22:
Hi John,

OK, I hadn't realised that the s-Meter cal also changed the graph. So this method is fine and you don't need to implement any other adjustment.

However the ability to select a rolling average / max / min line on the display would be useful :-)

Regards,

Martin