jks-prv/Beagle_SDR_GPS #28: Suggestion for users to be able to tag frequencies.

G8JNJ, 2016-10-18:
I think it would be useful if users were able to tag interesting frequencies to bring them to the attention of other users.

However I suspect that this could lead to abuse if it was not moderated in some way by the SDR admin.

Would it be possible to allow users to tag frequencies and add comments in the same way as the admin, but only display something like a question mark on a bright background colour, until the admin has opened up the tag and verified the comments, then changed the status so that it can be displayed normally.

Regards,

Martin - G8JNJ

jks-prv, 2016-10-18:
WebSDR has this concept of a user-contributed "log" where you can submit a description of what you're hearing. But those entries are not automatically turned into tags. In the wish list I already have this: "A mechanism for users to contribute to the kiwisdr.com update database". But you're asking for this on a per-kiwi basis which is also useful.

G8JNJ, 2016-10-18:
I had been thinking about a 'Master' frequency database so that individual SDR tags could be shared in some way.

It would be difficult to sort these into regions, but as the SDR's already have their co-ordinates defined I wondered if it could be done on a geographical basis. 

Tags could have a further field indicating where they originated from, this could be automatically entered by default from the GPS co-ordinates. 

Tags originating from other SDR's within a defined geographical radius cold then be sorted in a centralised database and made available for automatic download. 

Regards,

Martin - G8JNJ

G8JNJ, 2017-03-07:
Hi John,

I've been thinking some more about this.

Can I suggest that regardless of how any future shared database may be implemented I think it would be worthwhile adding some additional information to the data stored alongside tags on each SDR.

This would be the time and date the individual tag was added and also some reference to the SDR it was added to.

Then if at some time in the future, all individual tags from all the Kiwi's are uploaded to some sort of web based data base / search engine. It would be possible to see when the tag was added or updated and by who (which would give an indication to it's validity) and it would then also be possible to add a URL associated with the entry so that by clicking the user is taken to a web sdr where the signal has been heard.

Regards,

Martin - G8JNJ