jks-prv/Beagle_SDR_GPS #29: Suggestion for click to frequency dependent upon zoom level

G8JNJ, 2016-10-18:
I find that when zoomed out and tuning by clicking on the waterfall, I very often don't tune to the correct frequency.

Would it be possible to change the resolution of the tuning step that the SDR snaps to depending upon what level of zoom is being used.

For example most transmissions tend to be on frequencies that are multiples of whole KHz. 

When clicking on the waterfall at zoom level 0 to 2  1KHz resolution is OK, zoom level 3 to 4 500Hz, zoom level 5 to 6 10Hz and 1Hz steps at higher zoom levels.

Tuning using the + and - buttons and by direct frequency entry would continue to operate as before.

Regards,

Martin - G8JNJ

jks-prv, 2016-10-18:
Although not tied to the zoom level, I have something now called "rounded frequency set". See the "keyboard & mouse shortcuts" section of the quick start page: http://www.kiwisdr.com/quickstart/#id-user-shortcuts

G8JNJ, 2016-10-18:
Ah OK that works well.

Maybe the comment in the purple info box needs to be amended.

"Shift or ctrl-shift click in the waterfall to lookup frequency in online databases"

BTW I find this online database to be quite good for utility stations.

http://qrg.globaltuners.com/

Regards,

Martin - G8JNJ

jks-prv, 2016-10-18:
Oops, yes I forgot to update the info box when I changed the shift-click action. The Globaltuners database is searched when option/alt-shift-click is used instead of ctrl-shift-click. See: http://www.kiwisdr.com/quickstart/#id-user-shortcuts

G8JNJ, 2016-10-18:
Hi John,

Just a thought, maybe the user guide part of the quickstart document should be separated out and provided as a help menu ?

Thanks for all the positive stuff you have done so far :100:  

The Kiwi SDR continues to impress me :+1: 

Regards,

Martin - G8JNJ

jks-prv, 2016-10-18:
Yes, the "quick start" is pretty much out-of-control and needs to become a proper user page/manual.
Good idea about auto-linking some parts of the info into a help panel.

Well, overall the current Kiwi is a good start. But there are so many more things I'd like to add and fix.