jks-prv/Beagle_SDR_GPS #30: Possible bug LORAN_C extension

G8JNJ, 2016-10-18:
I've noticed that the Loran_C extension doesn't always sync to the correct time slot.

Stopping and restarting the extension a few times usually fixes it, but if the signal is not known it isn't really possible to tell if the timing is correct (or not).

Regards,

Martin - G8JNJ

jks-prv, 2016-10-18:
Not documented except in the Kickstarter update describing the extension, but you can click in the Loran-C display to shift the clicked point to the leftmost edge. For example with Anthorn, click just to the left of the master (group with 9 pulses) and it will shift to the beginning of the red "master" window. The slave group (8 pulses) will then appear properly somewhere in the yellow "Y" window.

A future extension version that has more time resolution could detect the master automatically.

G8JNJ, 2016-10-18:
Ah OK - worth knowing.

Not much Loran around these days :-(

Regards,

Martin - G8JNJ