jks-prv/Beagle_SDR_GPS #32: How can we help out?

rikvanriel, 2016-10-18:
I just set up my KiwiSDR recently, and set it up for public access. I love that there is a nice SDR with completely open source software.

I have worked on some other ham radio software in the past, and see a lot of great feature ideas for KiwiSDR in this issue list.

My question is, how can other software developers help out?

Have you considered setting up a development mailing list, or some other way for developers from outside the core team to get involved with KiwiSDR development?

I would be interested in working on demodulators, antenna switch control, and who knows what else, once I get more familiar with the code...

jks-prv, 2016-10-19:
I am desperately trying to wrap up shipping and taking care of the worst bugs so I can get back to documenting the extension API. At the same time I'm trying to make the support routines that extensions call a little cleaner. By far the worst part of writing an extension has been the HTML and Javascript code to operate the UI of the extension and the code required to move data back and forth between the server side written in C and the client side in Javascript. The support routines are supposed to make those tasks a bit easier.

There is already a forum over on the Valent F(x) site that might be suitable for development discussions: http://valentfx.com/vanilla/categories/kiwisdr-discussion

rikvanriel, 2016-10-19:
That is fair enough! The immediate things do need to happen first :)

I just want you to know that there will be other developers out there willing to help out with long term development, whenever you get around to that again. The KiwiSDR is (as far as I can tell) a great platform, and I suspect there are lots more things we could make it do with a development community.

Thank you for starting this project. I will check out the forum.

jks-prv, 2016-10-19:
I'm not sure I directly answered your question. I would appreciate and encourage help of any kind. Already a few people have sent patches. I'm a little embarrassed because much of the code was developed in a huge hurry and I cringe every time I think of people wadding through some of that code trying to understand it. I intend to make another pass and clean things up (and add comments!). So this issue is a bit of a double-edges sword I'm afraid..