jks-prv/Beagle_SDR_GPS #34: Saving AGC settings

G8JNJ, 2016-10-19:
Would it be possible to save the AGC settings as cookies ?

I find that setting the AGC is very beneficial in reducing the background noise level especially on the LF bands.

However the settings do not seem to be saved from session to session.

Also could the AGC defaults be set to something mid range. The default slope setting seems to be 0dB which means that the AGC doesn't do anything when initially enabled

My suggestion for default values would be:-

Threshold of -90dBfs  
Slope 6dB slope
Decay 1000mS 

Regards,

Martin - G8JNJ

jks-prv, 2016-10-19:
Sure, I can do that. I'll probably do the same thing as the mode buttons: if you shift-click on the button the AGC value will be restore to their default value.

I appreciate any suggestions for default values. I don't understand how all those AGC parameters are intended to work. The code is just lifted from CuteSDR. I had changed the threshold a while back from the default -120 to -100 but then noticed this made the volume of weak stations in CWN too low. Someone else suggested the current decay value to reduce noise "pumping" between speech pauses.

G8JNJ, 2016-10-22:
Hi John,

The values I suggested are a reasonable starting point for most configurations, especially all the SDR's that have very low signal levels to start with.

The threshold value should set the point at which the AGC action starts to kick in. Signals below this level are not modified, and signals above this level have their audio maintained at a near constant level, the maximum excursion of level being defined by the value of slope. Most AGC systems use somewhere between 6 to 10dB of slope.

The value of decay depends upon the type of modulation. CW normally has a short decay time <500ms and AM will normally have a longer delay perhaps 2 seconds or more. Most users will set the delay to their own liking. Hang helps to reduce the pumping effect that can sometimes be heard, by further modifying the time constants when a signal is varying in strength.

So when the AGC is operating correctly very low level signals will not be modified, but as the signal gradually increases in strength above the threshold value, progressively more attenuation will be applied.  This has the effect of improving the signal to noise ratio as the maximum signal level is defined by the strongest part of the received signal and the noise is kept well below this level.

However the numbering used for the threshold seems to work the opposite way around to what I would have expected, so I don't know whether it's correct or not. 

I would have thought that when no signal is present, as the threshold value was wound down towards the noise floor at approx -140dBfs then the gain reduction would have kicked in and reduced the audio level. 

However the opposite seems to happen and the audio level reduces as you raise the threshold value.

It maybe that I'm misunderstanding the convention that is being used, so I wouldn't worry too much about it.

Regards, 

Martin - G8JNJ

G8JNJ, 2016-10-22:
Ooops, pressed the wrong button

G8JNJ, 2016-11-06:
Hi John,

The current default setting for AGC is set to on.

However because of the default slope and threshold values it's not working properly, and is effectively the same as using a manual gain setting with AGC off.

To rectify this I have changed the AGC defaults on my WEB SDR.

openwebrx.js

Line 4407 var thresh = -100

Line 4417 var slope = 6

This seems to be a good default value as it reduces the background noise level when no signal is present but still allows the weaker signals to be heard.

Regards,

Martin

closed 2016-11-30 by jks-prv