jks-prv/Beagle_SDR_GPS #37: Audio FFT Extension - suggestion for timing adjustment

G8JNJ, 2016-10-23:
Hi John,

The audio FFT extension is very interesting.

I've tried using it to identify weak NDB's which transmit morse at regular intervals.

The tricky part is figuring out the correct repetition rate.

Would it be possible to click on the waterfall at two points and use that information to set the timing interval.

e.g. I can see the start of the morse ID so I click on that point then wait for the same sequence to start again and click on the same point of the sequence. This determines the cycle time and sets the integration time.

Would it also be possible to vary the speed of the FFT waterfall in order to cater for shorter of longer sequences. 

It would also be useful to be able to zoom in on the FFT waterfall in order to allow closer inspection.

Thanks,

Martin - G8JNJ

jks-prv, 2016-10-23:
Yup, like that idea. Another one I had was a slider that allows fine-tuning of the interval by re-summing the FFT samples when the interval is changed. So you would click as you describe to establish the rough cycle time, integrate for a few minutes, then fine-tune until those few minutes of summing line up. You could even have a mode where the individual FFTs are shown separately as an aid to fine-tuning in the proper direction to get the alignment correct.

For a strong enough signal you could even consider implementing an automatic fine-tune that attempts to detect the leading edge.

jks-prv, 2016-10-23:
One downside of this scheme is that changing the characteristics of the FFT isn't really possible since the FFT is being run at a fixed audio rate and size for the benefit of the FIR filter. The benefit is that you get it computationally for free. Not to say you couldn't write another extension that did a separate, arbitrary FFT on the audio stream with any parameters you wished.

G8JNJ, 2016-10-23:
Sounds good.

There's another interesting visual method of correlating weak repetitive signals such as NDB's which is incorporated into this software (21 day free trial). http://www.coaa.co.uk/ndbfinder.htm

Regards,

Martin - G8JNJ

jks-prv, 2016-10-23:
Yes, I've seen that software. Very impressive. Could have used that back in the 1980s when I did most of my serious NDB listening, lol.