jks-prv/Beagle_SDR_GPS #38: Waterfall frequency ruler fault

peejii73, 2016-10-24:
Hi,

It seems that at maximum waterfall zoom the frequency "ruler" at the top of waterfall shows potatoes, when the frequency is 1 MHz or more, and browser window is wide enough (> ~1880 pixels). Zooming one step back reverts proper display.

Best 73,
-petri OH1BDF

rikvanriel, 2016-10-24:
What exactly do you mean by "shows potatoes"?

I am looking at the kiwisdr waterfall in a >2000 pixel wide Chromium window right now, and things appear to behave like they should, except for the javascript engine sometimes being unable to keep up with all the waterfall pixels, and the audio looping for a little bit when I zoom in or out, or resize the browser window. That simply looks like an inefficiency in the web browser, though.

Is there a particular web browser that shows your issue?

peejii73, 2016-10-24:
Please have a look at a screenshot: https://dl.dropboxusercontent.com/u/4116321/potatoes.PNG

Chrome 54.0.2840.59 m & Firefox 49.0.1, running on W7/64. KiwiSDR FW is of version 1.16.

jks-prv, 2016-10-24:
Yes, I have known about this bug for a while (see in the bug list: "Reported case of repeating number text on frequency scale"). I have not been able to replicate it here so I can fix. I will try again..

icm7216, 2016-10-24:
Seems to be occurs it when using browser zoom function.

How to replicate this problem. Please try, set 67% of browser zoom-out, and to set max of kiwi-zoom, and then set frequency 1000 kHz.

icm7216, 2016-10-24:
In other case, seems to be occurs it when using the kiwi in a >1758 pixel wide display. 

How to replicate this problem. Please try, set 100% of browser zoom, and to set max of kiwi-zoom, and then set frequency 1000 kHz. 
next, set the "Select size" to 1758×500 into the Responsive Design Mode on Firefox. ( vertically size is any value.)

see: [Responsive Design Mode - Firefox Developer Tools | MDN](https://developer.mozilla.org/en-US/docs/Tools/Responsive_Design_Mode)

jks-prv, 2016-10-24:
Thanks everyone! I found the bug and it will be fixed in the next release.

closed 2016-10-24 by jks-prv