jks-prv/Beagle_SDR_GPS #41: kiwid stalls on sdr.hu registration

rikvanriel, 2016-10-25:
I have noticed occasional stalls of the waterfall and audio. I finally managed to grab a backtrace, and it turns out to be the sdr.hu registration that can take several seconds, stalling the kiwid process while waiting for the sdr.hu http request to complete (or the connection to be made?).

# gdb --batch -p `pidof kiwid` -ex "bt" -ex "quit"
__libc_do_syscall () at ../ports/sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:43
43	../ports/sysdeps/unix/sysv/linux/arm/libc-do-syscall.S: No such file or directory.
#0  __libc_do_syscall () at ../ports/sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:43
#1  0xb6c1650a in _IO_new_proc_close (fp=0x495d7d8) at iopopen.c:345
#2  0xb6c1d1c4 in _IO_new_file_close_it (fp=fp@entry=0x495d7d8) at fileops.c:171
#3  0xb6c1514c in _IO_new_fclose (fp=0x495d7d8) at iofclose.c:59
#4  0x0002ea98 in non_blocking_cmd (cmd=<optimized out>, reply=0x49ea468 "<!doctype html>\n<html>\n<head>\n\t<title>SDR.hu</title>\n\t<meta charset=\"utf-8\" />\n\t<meta http-equiv=\"X-UA-Compatible\" content=\"IE=edge\">\n\t

rikvanriel, 2016-10-25:
During these stalls, kiwid CPU usage drops to zero. It looks like maybe non_blocking_cmd is not as non blocking as hoped...

jks-prv, 2016-10-25:
How long are the stalls you are observing?

rikvanriel, 2016-10-25:
They last several seconds, long enough for the waterfall and sound buffers to run out. I think the longest I have seen is on the order of 10-15 seconds.

rikvanriel, 2016-10-25:
The stalls seem to correspond with failed sdr.hu registrations in /var/log/messages

Oct 25 04:42:34 kiwisdr kiwid: 2:12:15 0...      sdr.hu registration: FAILED n=0 sp=(nil)
Oct 25 04:42:34 kiwisdr rsyslogd-2007: action 'action 17' suspended, next retry is Tue Oct 25 04:44:04 2016 [try http://www.rsyslog.com/e/2007 ]
Oct 25 04:44:15 kiwisdr rsyslogd-2007: action 'action 17' suspended, next retry is Tue Oct 25 04:45:45 2016 [try http://www.rsyslog.com/e/2007 ]
Oct 25 04:45:01 kiwisdr kiwid: 2:14:42 0...      sdr.hu registration: FAILED n=1199 sp=(nil)

jks-prv, 2016-10-25:
Hmm. For the case where there is something in the reply buffer (n=1199 instead of zero), but the "UPDATE:" token isn't found as expected, it would be interesting to see what was in the buffer by using gdb. I don't log anything in this case (except saying "FAILED") because the buffer is likely to just contain a whole bunch of HTML.

rikvanriel, 2016-10-25:
Here is what I got back:

Oct 25 05:33:12 kiwisdr kiwid: 0:00:28 ....      sdr.hu registration: FAILED n=1199 sp=(nil) reply=<!doctype html>#012<html>#012<head>#012#011<title>SDR.hu</title>#012#011<meta charset="utf-8" />#012#011<meta http-equiv="X-UA-Compatible" content="IE=edge">#012#011#012#011<link rel="stylesheet" type="text/css" href="/gfx/font_bebasneue/stylesheet.css" />#012#011<link href="/js/bootstrap/css/bootstrap.min.css" rel="stylesheet">#012#011<link rel="stylesheet" type="text/css" href="/gfx/sdrhu.css" />#012#011<script type="text/javascript" src="/js/jquery/jquery-1.11.0.min.js"></script>#012#011<script src="/js/bootstrap/js/bootstrap.min.js"></script>#012#011<script type="text/javascript" src="/js/skroll/skrollr.min.js"></script>#012#011<script src="https://www.google.com/recaptcha/api.js"></script>#012#011<script type="text/javascript" src="/js/sdrhu.js"></script>#012#011#012#011<script type="text/javascript">var _paq=_paq||[];_paq.push(['disableCookies']);_paq.push(['trackPageView']);_paq.push(['enableLinkTracking']);

rikvanriel, 2016-10-25:
I guess github did its magic on the HTML tags. You were right though, sdr.hu simply returned a bunch of HTML, instead of the UPDATE: token you expected.

rikvanriel, 2016-10-25:
FWIW, even right after these failures, kiwisdr.surriel.com is still listed on sdr.hu.

ha7ilm, 2016-10-25:
The interesting parts of the response are under this part:
```
<a href="/" class="btn btn-success sdrhu-btn-bold"><span class="glyphicon glyphicon-share-alt"></span> Back to the front page</a><br /><br />
```
There should be a message over there.

ha7ilm, 2016-10-25:
sdr.hu/update can return the following different things:
* `UPDATE:ERROR (fields)`
* `UPDATE:ERROR (your IP does not match URL, 1.2.3.4 != 5.6.7.8)`
* `UPDATE:ERROR (apikey)`
* `UPDATE:ERROR (sql-update)`
* `UPDATE:SUCCESS (insert)`

It will not return a message if it is a GET query, otherwise it should.

rikvanriel, 2016-10-25:
It looks like wget fails with "Exit 4" (network failure), before receiving the entire text of the HTML page.

rikvanriel, 2016-10-25:
A simple fix for this issue may be to pass a shorter timeout on the wget commandline. A successful query seems to return in less than a second, and the waterfall & audio buffer appears to be on the order of 4 seconds, so maybe a 3 second wget timeout could be appropriate?

rikvanriel, 2016-10-25:
Running wget from the commandline various times seems to result in mostly successful registrations, but network issues (both no route to host, and incomplete timeouts) maybe 10-20% of the time.

ha7ilm, 2016-10-25:
Yeah, there seems to be some packet loss in your direction. 

```
PING kiwisdr.surriel.com (74.92.59.77) 56(84) bytes of data.
64 bytes from 74-92-59-77-NewEngland.hfc.comcastbusiness.net (74.92.59.77): icmp_seq=1 ttl=51 time=114 ms
64 bytes from 74-92-59-77-NewEngland.hfc.comcastbusiness.net (74.92.59.77): icmp_seq=3 ttl=51 time=113 ms
64 bytes from 74-92-59-77-NewEngland.hfc.comcastbusiness.net (74.92.59.77): icmp_seq=6 ttl=51 time=116 ms
64 bytes from 74-92-59-77-NewEngland.hfc.comcastbusiness.net (74.92.59.77): icmp_seq=7 ttl=51 time=115 ms
64 bytes from 74-92-59-77-NewEngland.hfc.comcastbusiness.net (74.92.59.77): icmp_seq=9 ttl=51 time=115 ms
64 bytes from 74-92-59-77-NewEngland.hfc.comcastbusiness.net (74.92.59.77): icmp_seq=12 ttl=51 time=111 ms
^C
--- kiwisdr.surriel.com ping statistics ---
15 packets transmitted, 6 received, 60% packet loss, time 14006ms
```

Another measurement showed around 8% packet loss over a few minutes, which was much better.

rikvanriel, 2016-10-25:
That turned out to be a local network issue unique to my kiwisdr. Other hosts on my network were not experiencing it. It appears to be an interaction between the bridging firewall sending redirects, and something on the kiwisdr. It should be fixed now.

ha7ilm, 2016-10-25:
I still see some pocket loss, but much better now.

jks-prv, 2016-10-25:
Let me know if you think I should change the timeout to make the wget more robust against marginal networks. I'm sure this issue will crop up again.

I had an interesting case yesterday where during the update process a wget of the Makefile (to check Kiwi version number) was failing because Telstra Australia, the ISP of this guy in Tasmania, was proxying the DNS for raw.githubusercontent.com someplace incorrect. Other things like sdr.hu registration worked fine. That one took us a while to find..

rikvanriel, 2016-10-26:
I figure there should be no harm to reducing that timeout to 3 seconds, which should be small enough to be barely noticeable to users if a network hiccup were to happen to some other kiwisdr.

jks-prv, 2016-10-26:
Okay, I'll do that.

jks-prv, 2016-11-03:
Well, sure enough, this has become an issue again. Someone has wget's to sdr.hu taking more than 3 seconds to complete. Don't know yet if it's a slow DNS or something. So it looks like having a 3 second timeout isn't going to work for him.

I've tried to recreate the original problem reported (Kiwi audio/waterfall stalls if the command run by non_blocking_cmd() hangs or takes a long time). But I just can't recreate it. In one experiment I used a command of "sleep 60; wget …" with non_blocking_cmd(). If you run the kiwi server like this "ng -stats" it will show you the runtimes of the internal private thread package. If I have a connection open to the Kiwi it takes about 45% of the runtime and the other 55% is idle time. You can see non_blocking_cmd() working properly when sdr.hu registration is active: waking up every 50 msec as it checks the shell command popen() for new output in a non-blocking manner. I need to run this experiment with a task that stalls waiting on network activity and maybe one running in a hard loop.

The only way the Kiwi process could be getting starved is if another Linux task was hogging the machine and the kernel time slicing wasn't giving sufficient time to the Kiwi server. How did you originally measure and decide that the Kiwi server was getting 0% cpu anyway?

rikvanriel, 2016-11-03:
I was logged in over ssh and running "top", and saw that kiwid was not using any CPU time at all. At the same time, kiwisdr had wget forked, and was reading from the wget pipe.

wget got a partial read from the web site, and kiwid was blocked on the pipe read.

jks-prv, 2016-11-04:
I looked more carefully at your original gdb backtrace and the hangup is in pclose/fclose(), not the O_NONBLOCK read(), which is strange. But then I googled "pclose hang" and there are all kinds of hits, some going back a decade or more. pclose() waits for the process on the other end of the pipe to exit. But you would think that wouldn't be a problem since the pipe read() has already signaled EOF with a zero length return. So maybe it gets hung in the network stack with weird TCP FIN handshake delays or something. There could be a million reasons.

I think I'll try putting the whole thing in a child process (like the update process does) that can delay as much as it wants without effecting the server.

jks-prv, 2016-11-05:
Okay, the v1.20 update going out later today has a fix for this. There is a new version of non_blocking_cmd() called non_blocking_cmd_child() that runs the whole thing in a Linux child process after a fork(). So any blocking, caused by pclose() or otherwise, shouldn't be an issue.

closed 2016-11-30 by jks-prv