jks-prv/Beagle_SDR_GPS #42: Spectrum window fixed, waterfall dynamic, not visually in sync.

ka7u, 2016-10-26:
The spectrum window is great for coarse tuning over a wide band, just jump to the peaked signal, but when zoomed in closer, the spectrum scale does not zoom dynamically to fit in the browser window, while the waterfall does. So the spectrum peaks are not aligned with the waterfall unless you take time to manually set the browser window size to match the fixed width of the spectrum display.

jks-prv, 2016-10-26:
Do you mean the spectrum window doesn't update like the waterfall when the browser window is resized? Because I see that now and that's definitely a bug (and easily fixed). But for me the spectrum always follows waterfall changes: zoom in/out, dragging left/right, etc.

jks-prv, 2016-10-26:
Rik has fixed the "spectrum not updating on window resize" part of this bug:
https://github.com/jks-prv/Beagle_SDR_GPS/pull/45

ka7u, 2016-10-26:
I forced a rebuild and now my spectrum window and waterfall stay synchronized when window resizes occur. Thanks for the fix!

closed 2016-10-27 by jks-prv