jks-prv/Beagle_SDR_GPS #48: Suggestion for expansion of existing SDR's

G8JNJ, 2016-10-30:
My SDR is now proving to be quite popular and my thoughts are turning towards how I could support more users.

Would it be possible to build some capability into the SDR so that if more Kiwi's and Beagles were deployed at a site they could be 'stacked' so that from a users perspective it just appeared as if they were just using the one SDR but with more receiver's ?

Regards,

Martin - G8JNJ

jks-prv, 2016-10-30:
Interesting idea. Yes, that should be possible.

G8JNJ, 2016-10-30:
OK that's useful to know.

It would be useful for modest increases in the number of users, but after a point it becomes an expensive way of adding capacity.

So how about something a bit more complicated. Maybe using the Kiwi as a headless server (or similar) to feed more powerful webserver hardware. Perhaps using the four receivers to stream say 8MHz of spectrum each ?

Regards,

Martin - G8JNJ

jks-prv, 2019-01-18:
A partial solution to this problem was implemented some time ago. You can configure multiple Kiwis to do "URL redirection" if all the channels of a Kiwi are busy. E.g. Kiwi #1 is set to redirect to Kiwi #2 if all the channels on Kiwi #1 are full. Only Kiwi #1 is listed on sdr.hu and the second Kiwi is used as needed.

See the connect tab of the Kiwi admin page.