jks-prv/Beagle_SDR_GPS #49: Suggestion for IBP beacon / WSPR propagation monitoring extension / new function

G8JNJ, 2016-10-30:
Hi,

I was thinking about how to maximise the use of the limited four receiver resources in each KiWi, especially when not being used by remote listeners.

I note that the KiWi currently has the IBP beacon tags changing in sequence with the beacon transmission schedule.

How feasible would it be to have the Kiwi scan the various IBP beacons on different bands and logging the results whenever there is a spare receiver that is not being used for other purposes. 

Maybe the WSPR extension could also scan following the WSPR hopping schedule when there is spare capacity on the Kiwi ? This would free up capacity in real time as it would not have individual users looged on for long periods of time just to run a WSPR sesion.

The IBP results could then be displayed on the individual KiWi as a form of real time propagation indicator.

Something like FAROS 

http://www.dxatlas.com/faros/

Regards,

Martin - G8JNJ

rikvanriel, 2016-10-31:
I would like background scanning of frequencies, too.

I would probably monitor jt65, jt9, psk31, and various marine modes like Navtex and DSC. 

On other words, background scanning should probably be configurable to some extent.

Maybe extensions could specify their own scanning sequence, and a main scheduler could iterate over the enabled ones. Then the kiwisdr admin can simply choose which background scanners to enable, and the extensions can log what they received, in addition to automatically sending spots to pskreporter, wsprnet, etc.

GeoNomad, 2017-09-02:
I have done a modification to scan the IBP beacons at the web interface. You can choose one beacon to monitor (the receiver follows the beacon and optionally mutes when the beacon is off), or cycle through all the beacons on all the bands over 15 minutes.

I attempted to feed the output to FAROS, but I think either the decimation causes it to not decode the CW, or the time delay puts it out of the decode range.

A method of logging the beacons is now on my list of projects.

Since this was done entirely at the web interface end, the method for inserting the js is a bit of a hack. See http://ve3sun.com/OpenWebRX/ for the scoop. I will fork this git later.

m1geo, 2018-07-30:
I would like to +1 this request. A feature to enable the WSPR receiver to run on all channels that are not in use. I.e., all receivers run WSPR. As users connect, the channels revert to being SDRs, and then return to WSPR receivers when the user disconnects.