jks-prv/Beagle_SDR_GPS #50: Question WRT WSPR bandwidth

G8JNJ, 2016-10-31:
Hi,

I notice that the WSPR extension defaults to a narrow CW bandwidth rather than the more usual 2.4KHz USB bandwidth.

WSPR has been optimised to expect a 2.4KHz wide block of signals and extract the WSPR signals from that  block. 

Has CW been chosen for any particular reason such as a need to reduce the processing required for WSPR decoding ?

Regards,

Martin - G8JNJ

jks-prv, 2016-10-31:
The WSPR code always searches for signals in a +/- 110 Hz range (220 Hz wide) from the center frequency. The decimated sample rate for the signal processing is slightly larger at 375 Hz. Even though the mode button says USB the filter bandwidth is adjusted to be 300 Hz wide placed on the WSPR center frequency (750 Hz). This narrow filter is used to improve the S/N ratio since the subsequent processing bandwidths are so narrow. I don't know where the idea that the optimal input bandwidth is 2.4 kHz would have come from.

G8JNJ, 2016-10-31:
Hi John,

I read it in the WSPR 2.0 User’s Guide by Joe Taylor, K1JT:

 "The SNR is computed "in a 2500 Hz reference bandwidth"

"Under most conditions a receiver bandwidth normal for voice SSB communication,
say 2.4 kHz, is a good choice. Narrower bandwidths are perfectly acceptable if you
have problems with strong signals well outside the 200 Hz WSPR passband. There
is no particular advantage in using a bandwidth as small as 200 Hz, however, since
WSPR does all necessary narrow-band filtering in software."

Someone else commented on wsprnet

You should use a bandwidth greater than 200Hz, if you look closely at the fine grain of the noise in the waterfall display at 200Hz bandwidth the filter causes smearing in the noise close to the WSPR band limits, due to the phase distortion caused by the sharp cutoff of the filter.

If you have a good quality 250 Hz bandwidth filter with low group delay in the passband it will reduce the phase distortion in the WSPR band and improve your decoder performance on stations close to the band edges, by placing the phase distortion just outside the WSPR band where it does no harm to the signals.

Setting the BW slightly wider than the 200Hz band is useful when some stations are slightly off frequency and just on the edge of the band.

Regards,

Martin - G8JNJ

jks-prv, 2016-10-31:
I think they are possibly talking about the limited choices of analog filters on non-DSP (or hybrid) radios. Although the comments about phase distortion and group delay problems of narrow digital filters are certainly valid.

All these things (except for the 375 Hz decimation processing bandwidth) could be made adjustable parameters on the admin/extensions/WSPR page instead of the hard-coded constants they are now in Beagle_SDR_GPS/extensions/wspr/wspr.h (i.e. FMIN/MAX, BW_MAX). BTW, you can edit these and just recompile to try out adjustments.

G8JNJ, 2016-10-31:
Hi John,

OK no problem, I was just interested to understand why the WSPR extension defaulted to a narrower bandwidth than I'd previously been used to.

I don't think it's a big issue, other than perhaps modifying the calculated S/N as a result of the narrower bandwidth. 

Regards,

Martin - G8JNJ

jks-prv, 2016-10-31:
I didn't know about the S/N calculation being dependent on the input bandwidth. But that's obviously important because the S/N is one of the parameters uploaded to the wsprnet.org reporting database.