jks-prv/Beagle_SDR_GPS #51: Suggestion - display local time on SDR WEB Page

G8JNJ, 2016-10-31:
Hi,

Whilst using other Kiwi SDR's on sdr.hu it sometimes surprises me that signals I'd expect to see are not present. Often this is simply because it's night time where the SDR is located.

Would it be possible to show the local time somewhere on the Kiwi header ? 

Perhaps with a coloured background or a sun / moon symbol (or similar to quickly indicate the time of day or night or when it's in the Greyline.

Regards,

Martin

jks-prv, 2016-10-31:
Yes, this is already on the bug list. Good idea about having a sun/moon symbol. I wanted to display local/UTC on the main page but also have an extension that would show a google map with a world day/night overlay like what wsprnet.org has (and sdr.hu needs, lol)

Display local time means I need to add setting the local timezone on the admin page which is a slight hassle. Debian on the Beagle is of course configured to operate with tz=UTC by default.

rikvanriel, 2016-10-31:
One question to ponder with this feature is "how little screen space can it occupy, while still showing what users want?"

I think a sun or moon symbol, with local time could be displayed next to the line containing:

"City name, Grid, [map]"

Having one drop-down box where the admin of the radio can select the time zone should be easy enough. Code to automatically guess the time zone from the grid locator would be more work, unless code to do that already exists somewhere under a license that allows us to use it....

G8JNJ, 2016-11-01:
Hi John & Rik,

OK all good ideas.

Just thinking out loud some other thoughts that crossed my mind were possibly changing the colour of say the frequency scale background to indicate the optimum frequencies for the time of day or perhaps a prediction of MUF or similar.

As Rik suggests the top line of the SDR would seem to be the best place for a clock.

I don't think you need a huge globe, just something that is big enough to provide an indication.

Maybe you could put a small animated globe on that line centered on the GPS position with a day / night shadow overlay ?

Regards,

Martin

closed 2017-05-14 by jks-prv