jks-prv/Beagle_SDR_GPS #54: Kiwi software should send/receive over VCP/VSP to 3rd party applications

jks-prv, 2016-11-03:
[This issue is from Ron, KA7U, that was originally submitted as part of another issue: https://github.com/jks-prv/Beagle_SDR_GPS/issues/19 Added here as a new issue for better visibility]

This image shows Fldigi monitoring the CW audio from the KiwiSDR web interface on an openSUSE OS. Pulse Audio is directing the audio connections and so Fldigi is the "recorder". The same could be done using virtual audio cables in Windows. If there were virtual serial ports such as com0com.exe provides in windows or SoCat provides in Linux, moving frequency and mode data to and from 3rd party applications such as Fldigi, then the KiwiSDR could be tuned (frequency and mode) from the 3rd party application and the 3rd party application could be tuned (frequency and mode) from the KiwiSDR web-browser waterfall input. This would allow for a great diversity of commonly available 3rd party applications to compliment KiwiSDR. That would be any instance of any KiwiSDR displayed on a web-browser using about any OS. While I'm at it, this link has very good overview information. It is written for HDSDR but the usage is similar. https://sites.google.com/site/g4zfqradio/hdsdr_digi-modes
[image]

bnordgren, 2019-01-18:
in another place, I noted a desire to export i/q data over the network with websockets. How does this issue relate to that?

jks-prv, 2019-01-18:
I think this issue is largely solved. We have the Python-based kiwiclient software (https://github.com/jks-prv/kiwiclient/tree/jks-v0.1) which demonstrates how to connect to the Kiwi API using web sockets to obtain audio and waterfall data. Either to processing in realtime or record to files for later analysis.

Then there is the CATSync application (https://catsyncsdr.wordpress.com) which has an embedded web browser such that CAT commands from ham rigs can effective "click the buttons" on the Kiwi user interface to make changes (frequency tuning etc.)