jks-prv/Beagle_SDR_GPS #56: Suggestion - keyword search for tags

G8JNJ, 2016-11-06:
Hi John,

I've now got a large number of tags setup on my SDR, however I'm finding it difficult to keep track of them all and remember where a particular service or broadcaster was located.

Would it be possible to include a simple keyword search with a next and back option that would sort the tags.

Regards,

Martin

jks-prv, 2016-11-06:
Excellent idea. I've updated the bug list. Maybe also only show the matching labels when a search is active. So when you're zoomed out the clutter of overlapping labels will be reduced. Search should also match the "notes" field so you can search e.g. "time station" and get multiple matches.

G8JNJ, 2016-11-07:
Hi John,

Good idea.

Is there any way to increase the variety of tags ?

I'd like to add some new ones to mark out broadcast stations and also Pirate / unlicenced users of the spectrum.

Regards,

Martin

jks-prv, 2016-11-07:
Yes. It's a fixed list now in the code, but no reason not to make it editable on the admin page like the band definitions will be.

jks-prv, 2021-02-27:
We have a "DX label filter" now. Select from the right-mouse-button menu or '@' shortcut key. Filters on the label "ident" and "notes" fields.

closed 2021-02-27 by jks-prv