jks-prv/Beagle_SDR_GPS #57: web page at :8073 refuses to connect

Maddox-zephyr, 2016-11-07:
I can ssh into the beaglebone and do ku or kd, but cannot get a web page.

Any hints?

Output of ms:

`Nov 07 01:04:39 kiwisdr kiwid[2387]: DEBIAN 8
Nov 07 01:04:39 kiwisdr kiwid[2387]: USE_SPIDEV
Nov 07 01:04:39 kiwisdr kiwid[2387]: LOAD_SPI = no
Nov 07 01:04:39 kiwisdr kiwid[2387]: Starting kiwid
Nov 07 01:04:40 kiwisdr kiwid[2387]: Start kiwid: OK
Nov 07 01:04:40 kiwisdr kiwid[2387]: Mon Nov  7 01:04:40 UTC 2016
Nov 07 01:04:40 kiwisdr kiwid[2393]: 0:00:00 ....      KiwiSDR v1.20 --------------------------------------------------------------------
Nov 07 01:04:40 kiwisdr kiwid[2393]: 0:00:00 ....      compiled: Nov  7 2016 00:44:11
Nov 07 01:04:40 kiwisdr kiwid[2393]: 0:00:00 ....      reading configuration from file /root/kiwi.config/kiwi.json
Nov 07 01:04:40 kiwisdr kiwid[2393]: 0:00:01 ....      serial number from EEPROM: 1389
Nov 07 01:04:40 kiwisdr kiwid[2393]: 0:00:01 ....      reading configuration from file /root/kiwi.config/dx.json
Nov 07 01:04:40 kiwisdr kiwid[2393]: 0:00:01 ....      883 dx entries
Nov 07 01:04:40 kiwisdr kiwid[2393]: 0:00:01 ....      listening on port 8073 for "openwebrx"
Nov 07 01:04:40 kiwisdr kiwid[2393]: 0:00:01 ....      webserver for "openwebrx" on port [::]:8073
Nov 07 01:04:41 kiwisdr kiwid[2393]: 0:00:01 ....      ### using SPI_DEV
Nov 07 01:04:42 kiwisdr kiwid[2393]: 0:00:02 ....      FPGA version 1
Nov 07 01:04:42 kiwisdr kiwid[2393]: 0:00:02 ....      CFG read /root/kiwi.config/kiwi.json: sdr_hu_register = false
Nov 07 01:04:45 kiwisdr kiwid[2393]: 0:00:05 ....      (0 users)
Nov 07 01:04:45 kiwisdr kiwid[2393]: 0:00:05 ....      UPDATE: scheduled
Nov 07 01:04:45 kiwisdr kiwid[2393]: 0:00:06 ....      UPDATE: checking for updates
Nov 07 01:04:46 kiwisdr kiwid[2393]: 0:00:06 ....      UPDATE: version 1.20 is current
Nov 07 01:04:46 kiwisdr kiwid[2393]: 0:00:07 ....      DDNS: private IPv4 <192.168.1.13> 0xc0a8010d /24 0xffffff00
Nov 07 01:04:46 kiwisdr kiwid[2393]: 0:00:07 ....      DDNS: public ip 47.185.229.231
`

Maddox-zephyr, 2016-11-07:
hmmm log output is hard to read, try again:

Nov 07 01:04:39 kiwisdr kiwid[2387]: DEBIAN 8
Nov 07 01:04:39 kiwisdr kiwid[2387]: USE_SPIDEV
Nov 07 01:04:39 kiwisdr kiwid[2387]: LOAD_SPI = no
Nov 07 01:04:39 kiwisdr kiwid[2387]: Starting kiwid
Nov 07 01:04:40 kiwisdr kiwid[2387]: Start kiwid: OK
Nov 07 01:04:40 kiwisdr kiwid[2387]: Mon Nov  7 01:04:40 UTC 2016
Nov 07 01:04:40 kiwisdr kiwid[2393]: 0:00:00 ....      KiwiSDR v1.20 --------------------------------------------------------------------
Nov 07 01:04:40 kiwisdr kiwid[2393]: 0:00:00 ....      compiled: Nov  7 2016 00:44:11
Nov 07 01:04:40 kiwisdr kiwid[2393]: 0:00:00 ....      reading configuration from file /root/kiwi.config/kiwi.json
Nov 07 01:04:40 kiwisdr kiwid[2393]: 0:00:01 ....      serial number from EEPROM: 1389
Nov 07 01:04:40 kiwisdr kiwid[2393]: 0:00:01 ....      reading configuration from file /root/kiwi.config/dx.json
Nov 07 01:04:40 kiwisdr kiwid[2393]: 0:00:01 ....      883 dx entries
Nov 07 01:04:40 kiwisdr kiwid[2393]: 0:00:01 ....      listening on port 8073 for "openwebrx"
Nov 07 01:04:40 kiwisdr kiwid[2393]: 0:00:01 ....      webserver for "openwebrx" on port [::]:8073
Nov 07 01:04:41 kiwisdr kiwid[2393]: 0:00:01 ....      ### using SPI_DEV
Nov 07 01:04:42 kiwisdr kiwid[2393]: 0:00:02 ....      FPGA version 1
Nov 07 01:04:42 kiwisdr kiwid[2393]: 0:00:02 ....      CFG read /root/kiwi.config/kiwi.json: sdr_hu_register = false
Nov 07 01:04:45 kiwisdr kiwid[2393]: 0:00:05 ....      (0 users)
Nov 07 01:04:45 kiwisdr kiwid[2393]: 0:00:05 ....      UPDATE: scheduled
Nov 07 01:04:45 kiwisdr kiwid[2393]: 0:00:06 ....      UPDATE: checking for updates
Nov 07 01:04:46 kiwisdr kiwid[2393]: 0:00:06 ....      UPDATE: version 1.20 is current
Nov 07 01:04:46 kiwisdr kiwid[2393]: 0:00:07 ....      DDNS: private IPv4 <192.168.1.13> 0xc0a8010d /24 0xffffff00
Nov 07 01:04:46 kiwisdr kiwid[2393]: 0:00:07 ....      DDNS: public ip 47.185.229.231

jks-prv, 2016-11-07:
You can get some additional debugging messages by running the server in the foreground. Do a "mst" to stop the background running. And then 'k' to start in the foreground. Then try connecting from your browser. You should see some messages if there is any attempt connecting at all. Is your machine with the browser on the same local network as the Kiwi? (i.e. 192.168.1.0/24) Or are you trying to access the Kiwi's public address 47.185.229.231?

Maddox-zephyr, 2016-11-07:
Well this is embarrassing - I had entered 192.168.1.13/8073 instead of 192.168.1.13:8073

All works well.

closed 2016-11-07 by jks-prv