jks-prv/Beagle_SDR_GPS #58: Frequency input: accept also comma as a decimal separator

peejii73, 2016-11-07:
Some countries use comma as a decimal separator (for instance Nordic ones, DK/SV/NO/FI). This is often effective in numeric keypad mapping, where decimal separator reflects to system locale.

Would it be a big effort to accept both period and comma as decimal separators? Additionally, if system locale is available at browser / JS level, maybe it could be used to parse the F input (and when presenting frequencies, though this isn't that important..) correct way?

73
-petri OH1BDF

jks-prv, 2016-11-07:
Good idea. I've never looked into it but I'm sure Javascript has support for internationalization at some level.

peejii73, 2016-12-07:
Oh javascript! Not so obvious to a low level C / assembly guy :)

```
diff --git a/web/openwebrx/openwebrx.js b/web/openwebrx/openwebrx.js
index dfb1f5a..e9dd66f 100644
--- a/web/openwebrx/openwebrx.js
+++ b/web/openwebrx/openwebrx.js
@@ -3012,7 +3012,7 @@ function freqset_complete(timeout)
        var obj = html_id('id-freq-input');
        //console.log("FCMPL t/o="+ timeout +" obj="+(typeof obj)+" val="+(obj.value).toString());
        if (typeof obj == "undefined" || obj == null) return;           // can happen if SND comes up long before W/F
-       var f = parseFloat(obj.value);
+       var f = parseFloat(obj.value.replace(",", "."));
        //console.log("FCMPL2 obj="+(typeof obj)+" val="+(obj.value).toString());
        if (f > 0 && !isNaN(f)) {
                freqmode_set_dsp_kHz(f, null);
```

jks-prv, 2016-12-09:
Okay, this will be in the next release.
Thanks!

closed 2016-12-22 by jks-prv