jks-prv/Beagle_SDR_GPS #60: Link to external (IRC) chatroom

linkfanel, 2016-11-14:
Chatting is nice, and I've taken notice of the wish list item:

> Chat? (that's been kind of a mess on WebSDR hasn't it?)

I would like to point out a few thoughts about it.

I think your project is great, and the future of WebSDRs, but I don't think you should try to add every suggested feature into it. In my opinion you shouldn't try to do everything within KiwiSDR. You should keep a modular design for the core jobs of KiwiSDR, maybe implement an extension framework for peripheral jobs - that's ambitious :) - but let other separate tools handle what are clearly other jobs.

As you've pointed out, chat has been a mess on the utwente WebSDR. Probably because WebSDR administration is very different from chat administration, the operator doesn't have the time to do it correctly, the architecture, functionality and support behind the chat are subpar compared to what you would expect from real chatrooms, etc. In my opinion it should have never been attempted to begin with.

On KiwiSDR, these problems would be compounded with the fact that with only 4 users maximum, you wouldn't have many people to chat with.

Instead, I suggest that in the user panel, you simply add a link to an external chatroom, which would be configurable by the administrator. IRC sounds like a great option for that. It could just be IRC chatroom information, an irc:// URL if that's universally supported, and/or a link to a WebIRC in a new browser tab for those without an IRC client. It could also be configured to link to an XMPP chatroom or whatever else it is that people want to use to chat.

The link could point to a default KiwiSDR listening IRC chatroom, which would then be shared by KiwiSDR users from any instance, and would benefit from the network effect. Administrators could customize it to a regional chatroom too, or a personal one, or disable the link if they so desire.

The best about this approach, especially compared to a chatbox, is that it is **very** easy to implement; about a one-liner for a static link, not much more for a configurable one. You don't have to administrate the chatroom yourself, you could even point to an existing SWL one.

I hope this helps and we can see this soon!

jks-prv, 2016-11-14:
Great points Pierre. Thank you.

I should probably make this more clear, but the bug / wish list is also meant to be an exhaustive list of all the suggestions made so people don't keep suggesting the same ideas over and over. I sometimes spend hours responding to all the emails each morning, lol. Maybe there needs to be an new icon on the list that means "will not do".

Completely agree about a single shared IRC being much more useful than attempting an integrated chat per-receiver. I've never used any chat system, so suggestions about possible IRC hosts would be most welcome.

The other thing I really need to do is get the blog function off of Kickstarter updates and move it someplace else. There are a lot of new customers now who are not getting email notification of new KS updates and would have to check it manually (if they know about it at all).

linkfanel, 2016-11-14:
> I sometimes spend hours responding to all the emails each morning, lol. Maybe there needs to be an new icon on the list that means "will not do".

Ah yes, maybe! I can feel your efforts sorting through all of these.

I would suggest linking to the freenode IRC network. It's popular and has good staffing and support. As for the chatroom, you could go for #kiwisdr, or something more general about shortwave listening, or whatever you want and is available. You could link to `irc://chat.freenode.org:6667/kiwisdr` You could also create some DNS that redirects to freenode and link to `irc://irc.kiwisdr.com:6667/kiwisdr` or `irc://irc.sdr.hu:6667/kiwisdr` That way if you ever move to another IRC network you can just change the redirection, and it's easier to remember for some people.

Those are URLs that will open into an IRC client if it's installed. Aside from that, I would suggest for WebIRC options https://kiwiirc.com/client/chat.freenode.net/#kiwisdr or https://webchat.freenode.net/?channels=%23kiwisdr

G8JNJ, 2016-11-15:
Hi,

I think this is a good idea.

Quite often when I hear a strange signal or see someone else using my SDR on an 'odd' frequency I'd really like to be able to ask others what the signal is, or what it is they are listening to.

It would be good to be able to exchange Kiwi URL's with tuning information etc.

Regards,

Martin

closed 2017-05-14 by jks-prv