jks-prv/Beagle_SDR_GPS #62: PANIC: "jsmn_parse" (cfg.c, line 496)

ka7u, 2016-11-18:
After occasionally adding stations to the dx.json file the reload crashed with:
Nov 17 23:55:01 kiwisdr kiwid: 0:00:00 ....      compiled: Nov  9 2016 04:28:49
Nov 17 23:55:01 kiwisdr kiwid: 0:00:00 ....      reading configuration from file /root/kiwi.config/kiwi.json
Nov 17 23:55:01 kiwisdr kiwid: 0:00:01 ....      serial number from EEPROM: 1361
Nov 17 23:55:01 kiwisdr kiwid: PANIC: "jsmn_parse" (cfg.c, line 496)
Nov 17 23:55:13 kiwisdr kiwid: 0:00:00 ....      KiwiSDR v1.22

The last line entered that seems to have generated the error is shown in **bold** below. I cut the newly added stations list as shown below, rebooted, and all is well again. I pasted it back in and it crashed again. Took it out and all is well. How do you restart kiwisdr from the command line?

,[580.00,"AM","KIDO","5KW,Nampa,ID",{"WL":1}]
,[610.00,"AM","KONA","5KW,Richland,Pasco,Kennewick,WA",{"WL":1}]
,[630.00,"AM","KFXD","5KW,Boise,ID",{"WL":1}]
,[670.00,"AM","KBOI","50KW,Boise,ID",{"WL":1}]
,[710.00,"AM","KIRO","50KW,Seattle,WA",{"WL":1}]
,[730.00,"AM","KNFL","15KW-D,500W-N,Boise,ID",{"WL":1}]
,[760.00,"AM","KDSP",50KW-D,1KW-N,Thornton,CO",{"WL":1}]
,[770.00,"AM","CHQR","50KW,Calgary,Alberta",{"WL":1}]
,[770.00,"AM","KTTH","50KW-D,5KW-N,Seattle,WA",{"WL":1}]
**,[780.00,"AM","KKOH","50KW,Reno,NV",{"WL":1}]**
,[790.00,"AM","KSPD","1KW-D,61W-N,Boise,ID",{"WL":1}]
,[890.00,"AM","KYWN","50KW-D,250W-N,Meridian,ID",{"WL":1}]
,[1060.00,"AM","KBGN","10KW,Caldwell,ID",{"WL":1}]
,[1140.00,"AM","KGEM","10KW,Boise,ID",{"WL":1}]
,[1350.00,"AM","KTIK","5KW-D,600W-N,Nampa,ID",{"WL":1}]
,[1380.00,"AM","KKOO","15KW-D,1KW-N,Ontario,OR",{"WL":1}]
,[1490.00,"AM","KCID","1KW, Caldwell,ID",{"WL":1}]
,[1650.00,"AM","Weiser Highway Advisor","10W,Ontario,OR",{"WL":1}]

ka7u, 2016-11-18:
I've attached the user.log if that will help to see how I messed it up.

[user.log.zip](https://github.com/jks-prv/Beagle_SDR_GPS/files/598760/user.log.zip)

jks-prv, 2016-11-18:
The offender is the missing double quote on this line:
,[760.00,"AM","KDSP",50KW-D,1KW-N,Thornton,CO",{"WL":1}]

Someday we'll be able to upload from a .csv file and avoid this kind of thing. I should also probably change the JSON parser to print out the line number of where the problem is. That parser is third-party open source, but pretty simple.

You can restart the background Kiwi server daemon with the "ku" (kiwi up) command.

ka7u, 2016-11-18:
John, Thank you for taking the time to find my fault. ku will be useful for working with the tags.

jks-prv, 2016-11-18:
Okay, the next release will log something like this:
```
Fri Nov 18 07:17:41 2016 0:00:00 ....      cfg_parse_json: file /root/kiwi.config/dx.json line=10 pos=612 tok=79
Fri Nov 18 07:17:41 2016 0:00:00 ....      invalid character inside JSON string
Fri Nov 18 07:17:41 2016 0:00:00 ....      PANIC: "jsmn_parse" (cfg.c, line 496)

```
So at least you get a hint about which file & line has the problem.

closed 2016-11-18 by jks-prv

ka7u, 2016-11-20:
This being closed I don't know if you will get to read it or not, but want to say that your log error line number is a great enhancement for a guy like me. I've already used it since the recent upgrade to 1.23, Thank you.

jks-prv, 2016-11-20:
Yes, I get notifications about comments on closed issues. Useful since we've had a couple bugs resurface after they've been "fixed", lol.