jks-prv/Beagle_SDR_GPS #63: Waterfall stopped working on Firefox [i386 Linux]

drbos, 2016-11-19:
I noticed today that the KiwiSDR waterfall no longer displays on my small laptop, an Acer Aspire One i386 running Firefox 43.0 under Mint Linux.  None of the current KiwiSDRs listed on sdr.hu displays a waterfall.  Both of the RTL-SDR RPI2 sites I tried from there work fine.   

Seems to be a result of a change introduced in a recent KiwiSDR update.

jks-prv, 2016-11-19:
Before (re)loading a Kiwi site, on Firefox can you do "Tools > Web Developer > Web Console" first? And then see if there are any error messages (typically in red) when you load the page.

Also (and this is almost an Internet meme by now) try clearing your cache first. We have noticed that sometimes someone will connect using an old version of the javascript code. This is not supposed to be possible since the browser should detect that whatever it has cached is out-of-date w.r.t. what the Kiwi is serving up. This tends to break things when the protocol for the waterfall or audio streams is changed as we tinker with things. We try to remain backward-compatible for this reason. But we are also trying to clean up the code which is a bit of a mess.

drbos, 2016-11-19:
Thanks for your swift response.

Clear Cache has no effect.  
Web Console shows multiple recurring errors "TypeError p is undefined" 
in openwebrx.js:
&nbsp;&nbsp;shift_canvases()
&nbsp;&nbsp;&nbsp;&nbsp;waterfall_add()
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;waterfall_dequeue()

icm7216, 2016-11-19:
Firefox clearing cache (override cache) is `Ctrl` + `F5` or, `Ctrl` + `Shift` + `R`

See also: [Keyboard shortcuts - Forefox](https://support.mozilla.org/en-US/kb/keyboard-shortcuts-perform-firefox-tasks-quickly)

The latest version works well with Firefox 50.0. Firefox 43.0 is too old and has security problems. Please update to the latest version.

See also: [Update Firefox to the latest version](https://support.mozilla.org/en-US/kb/update-firefox-latest-version)

jks-prv, 2016-11-19:
Well, this is a tough one. There shouldn't be any cases where 'p' is undefined in that code unless the DOM timing of that particular FF version is doing something funny. I'd like to know if this problem goes away with an update of FF to version 50.x Could you tell me which line number in openwebrx.js is shown for that error? Thanks.

drbos, 2016-11-19:
I notice a similar problem on a work (Windows) PC and it went away when I changed to Firefox 50.0

Line numbers are:
shift_canvases() openwebrx.js:1969
waterfall_add()  openwebrx.js:2264
waterfall_dequeue() openwebrx.js:2455

FWIW Chrome on the same laptop shows the same problem.  Version is:
Google Chrome	45.0.2454.101 (Official Build) (32-bit)

jks-prv, 2016-11-19:
I added some code to prevent the error from being displayed, but you would still probably get no waterfall. The oldest browser I could find was FF 48 and Chrome 49 and they worked fine. So I'm going to mark this "won't fix" since I don't have a way to track down the problem.

jks-prv, 2016-11-19:
I just had a report of this happening with FF 50.0. So re-opening for now.

drbos, 2016-11-19:
Here's a kludge that works.  
Disable function shift_canvases at about line 1965 in openwebrx.js with:
<pre>
function shift_canvases()
{
	// shift the canvases downward by increasing their individual top offsets
	canvases.forEach(function(p) 
	{
		p.style.top = (p.openwebrx_top++).toString()+"px";
	});
	<b>return;	// >>>>>>>>>>>>DRB<<<<<<<<<<<<<<< </b>
	// retire canvases beyond bottom of scroll-back window
</pre>

It restores the waterfall display and doesn't obviously break anything else.

drbos, 2016-11-19:
There was in fact a noticeable problem with the ugly fix above -- a change in waterfall displayed frequency produced an ugly stuttering of the waterfall until the screen filled.  
The fix below works, avoids that problem, and is a little more subtle:
<pre>
       canvases.forEach(function(p)
        {
                p.style.top = (p.openwebrx_top++).toString()+"px";
        });
        // retire canvases beyond bottom of scroll-back window
        var p = canvases[canvases.length-1];    // last not including the phantom
        <b>if (p === undefined) {return;}  // >>>>>>>>>>>>DRB<<<<<<<<<<<<<<< </b>
        if (p.openwebrx_top >= waterfall_scrollable_height) {
</pre>

jks-prv, 2016-11-19:
I also added a similar change. If this fixes your problem then I think I now understand why this would have prevented the waterfall from appearing at all. I still don't understand the root cause -- why canvases[] ends up being empty when shift_canvases() is called, but here is why it kept the waterfall from appearing. In the v1.14 release a different audio/waterfall sync mechanism was introduced. Part of that included this code:
```
		waterfall_add(waterfall_queue.shift());
		waterfall_queue_seq.shift();
		waterfall_queue_spacing.shift();

```
The problem with that code is if the waterfall_add() routine generates an un-caught exception then the three queues don't get shift()'d equally (the seq and spacing queues would be one entry greater). This would confuse the sync code and prevent the waterfall from ever appearing again. A better solution is to use a single queue containing objects with all three data items. This change will go out in the huge update scheduled for later today.

icm7216, 2016-11-19:
I could not reproduce this problem in my environment. Waterfall works fine in my environment. 
see below:

### ubuntu 16.04 LTS (64-bit) on Virtual Box 5.1.8 for Windows

*   Firefox 50.0 (64-bit)
*   Firefox Developer Edition 51.0a2 (64-bit)
*   Firefox Nightry 53.0a1 (64-bit)

### Windows 10 Pro 64-bit

*   Firefox 50.0 (32-bit WOW64)
*   Firefox Developer Edition 51.0a2 (32-bit WOW64)
*   Chrome 54.0.2840.99 m

jks-prv, 2016-11-19:
The changes going out in the v1.23 update later today will at least prevent the waterfall from stopping, even if the underlying problem is not fully understood.

icm7216, 2016-11-19:
I also tried it with Firefox 43.04 (64-bit),  on ubuntu 16.04 LTS (64-bit) on Virtual Box 5.1.8 for Windows.

Waterfall works fine.
[image]

drbos, 2016-11-20:
KiwiSDR v1.23 has killed the waterfall again.  
The Javascript error is "canvas is undefined" at openwebrx.js:2239
Adding
<pre>
  if (canvas === undefined) { return; }
</pre>
ahead of that line gets rid of the error messages but still no waterfall -- canvas is apparently always undefined.

jks-prv, 2016-11-20:
But you mean with old browsers, right? Is anyone having problems with v1.23 on an up-to-date browser?

drbos, 2016-11-20:
I think the problem is not "old browser" but Firefox/Chrome  on a 32-bit system.
I get the same problem with  32-bit Firefox v50.0

jks-prv, 2016-11-20:
I'd be interested to know if the change I just checked in makes any difference. Thanks.
To recompile:
cdp
m git
m
mi
ku

drbos, 2016-11-20:
Yes!  Works great.
Thanks -- Ross

jks-prv, 2016-11-20:
Well, I don't get it. I must be missing something subtle. Here is the code in simplified javascript. The only change is whether waterfall_add() gets the current canvas from the front of the wf_canvases[] array or from a global var. The Openwebrx code also uses a global var for this, but wf_canvases[] has to be valid too because it is used successfully later on in waterfall_add() by wf_shift_canvases() and add_canvas(). wf_canvases[0] should always hold the same value as wf_cur_canvas. Anyone see a problem here? It's obviously not 100% wrong since it works on a lot of browser / system combinations.
```
function init_canvas_container
	add_canvas()	// add first one

waterfall_queue = []

function waterfall_add_queue		// called when waterfall data received from network
	waterfall_queue.push(data)		// add to tail

setInterval(waterfall_dequeue, 45)		// call at 22 Hz

function waterfall_dequeue
	while (condition)
		data = waterfall_queue[0]	// take front
		waterfall_queue.shift()		// remove from front
		waterfall_add(data)

function waterfall_add
	// *** these two lines were the only change
	//canvas = wf_canvases[0]		// most recently added from front
	canvas = wf_cur_canvas			// most recently added (from global var)

	if (canvas == null) return
	wf_shift_canvases()
	if (condition) add_canvas()

wf_canvases = []
wf_cur_canvas = null

function add_canvas()
	new = ...
	wf_canvases.unshift(new)	// add to front
	wf_cur_canvas = new			// also save in global var

function wf_shift_canvases
	if (condition)
		wf_canvases.pop()	// remove from tail

```

closed 2016-11-30 by jks-prv