jks-prv/Beagle_SDR_GPS #68: Default user is root

Tenacious-Techhunter, 2016-11-30:
A non-root default user is more (if only marginally more) secure than having the default user be root.

jks-prv, 2016-11-30:
There is an open bug (http://www.kiwisdr.com/bugs/) to try and run the server sandboxed, chrooted, or something. There are issues with device access, mmap() etc.

elfchief, 2016-11-30:
Do you have specific details on what needs to be accessed, and how? I'd potentially be interested in working on making this (and sandbox-related) changes if details are available.

elfchief, 2017-01-05:
Ping?

linkfanel, 2017-01-06:
An easy thing that can be split off is automatic updates. I can't see reasons for the current way it's done, but I see reasons against. Updates are driven from the server binary, involve fetching Makefile and source code over more or less secure channels, running the Makefile as root, compiling the source code as root and in place in /root. Compiling as root is always discouraged, but here the injection danger is big.

Since KiwiSDR boards are all the same, I think the compiled binaries can be the same everywhere? I think release builds should be run only on one build box, then the binaries packaged (even a tar archive will do), and uploaded to a master server. Automatic updates can be a cron script querying the master server for available updates, downloading the new package and installing it: for a tar archive, that would be decompressing the files in a per-version directory, then instantly changing a symbolic link pointing to installed versions from the previous to the new one and restarting the server.

All the complexity is removed from the server binary, the downtime is eliminated with only a restart left, the exposure of the root user is reduced to downloading and extracting files and restarting the server, and the walls against compromission through automatic updates are raised.

antonjan, 2017-01-06:
Hi

I would prefer the source code compiler on kiwi so I can change it for my
own custom configuration.

linkfanel, 2017-01-06:
Hello, yes of course it's nice to do personal builds, I don't mean that should go away. There would still be scripts to fetch the code and build it, and these scripts could still be called manually on individual KiwiSDR boards for custom builds. But it's good to clearly separate what doesn't need to be jumbled together. Compiling functionality is important, but not for automatic updates as root within the server binary :)

closed 2017-06-13 by jks-prv

rangerdude, 2024-04-16:
How is this closed as a "non-issue" Its 2024  you have had time to fix this.