jks-prv/Beagle_SDR_GPS #72: Is it possible to zoom in even further ?

G8JNJ, 2016-12-11:
Hi John,

I've recently been trying to identify some interference on the 475KHz band.

I've noticed that on the LF and NDB bands or when looking at narrow shift FSK / CW it's not possible to zoom into the frequency range sufficiently to be able to pick out individual carriers. I have to feed the SDR audio into Spectrum lab or similar to be able to do this.

Would it be possible to increase the Zoom in by a factor of 10, or is there a limitation due to the hardware / number of FFT bins ?

Regards,

Martin

backburn, 2017-01-08:
Second this.

jks-prv, 2017-01-08:
There are some limitations. Basically you just sort of have to try it and see if it works and you can still meet timing under a 4 channel load. There are already some heroics taken to make it work (fancy scheduling etc.)

The waterfall FFTs are 8k points now and take less than the 8 msec latency requirement of servicing the audio channel buffers. If you start doing larger FFTs to increase the resolution you get into trouble. The FFTW package I use is a "black box". Yes, can be compiled from source code. But I'm not really interested in trying to figure out how to insert a preemption-point in the inner loop of that code (which has the crap optimized out of it). It would destroy the FFT efficiency anyway. And I've avoided the idea of doing asynchronous task preemption with interrupts or signals or something because of the grief that would bring to the codebase (you'd have to start using locks etc.)

The other thing to do is increase the upper limit of the waterfall FPGA zoom-DDC ahead of the FFT. I'm not sure what would happen to things like spurious responses if you increased by 10x (there is already a spurious response problem I need to work on at some point). Increasing the DDC decimation doesn't require any additional FPGA memory, which is good because we're out of it. But it does require more gates although there should be plenty.

G8JNJ, 2017-01-09:
Hi John,

Hmm, it's a bit more complicated to implement than I'd hoped. 

If it's a trade between performance or being able to zoom in further then I'd stick with the present performance.

Some suggestions.

How is the waterfall in the WSPR extension implemented ? 

Could the existing 'Integrate' waterfall be enhanced further ? Perhaps it could be set so that it doesn't overlay consecutive scans and also has a frequency scale or grid ?

Perhaps it could be set to provide a longer horizontal scan with a vertical frequency scale ?

Regards,

Martin

jks-prv, 2017-01-09:
I tried adding another 4 levels of zoom yesterday. It sort of works, but uncovered a whole bunch of problems that need to be fixed. I didn't try it with 4 channels.

WSPR does its own separate, very small FFT on the audio IQ stream. The bandwidth is small so the number of FFT points can be small and doesn't have much impact on the system realtime issues.

The "integrate" extension re-uses the data from the FFT done as part of the audio passband FIR filter. Although we later discovered that this FFT doesn't have the usual "window function" applied ahead of FFTs used in displaying data, so it isn't ideal (there are artifacts on the left/right display ends). Because of that the bug list mentions that there should really be a separate audio channel FFT designed specifically for display with all the resolution and controls you would want (Martin you've mentioned some of those in the issues list previously and also above). This would be separate from the waterfall of course by being an extension.

G8JNJ, 2017-01-09:
Hi John,

OK, Understood. 

Maybe an extension that provides a waterfall function like Argo that is just contained within the RX pass band could be the way to go.

Regards,

Martin

jks-prv, 2017-01-14:
There is a tentative version for testing that supports two more zoom levels (z12, z13).
To get it login to your Beagle as root, "cdp" then "up".

Still some strange behavior with the yellow passband graphic now that is can occupy even more of the waterfall width.

G8JNJ, 2017-01-14:
Hi John,

OK did that now I can't see anything on the SDR pages.

Chrome  Console view says 'Failed to load resource: the server responded with a status of 404 (Not Found)' 

I can still login as Admin.

Any ideas ? Can I recover easily ?

Regards,

Martin

jks-prv, 2017-01-14:
Try the process again. There was a bug with the server running as a background daemon.

G8JNJ, 2017-01-15:
Hi John,

OK it worked that time :-)

It looks very good. I can now separate NDB morse ID's from each other and I can now also see the Phase modulated data that is carried on the BBC R4 transmitter on 198KHz in quite good detail.

[image]

I have so far not noticed any of the other artifacts that you have described.

Regards,

Martin

G8JNJ, 2017-01-15:
Hi John,

The zoom function is very good. I can now see the signals in a lot more detail and it's possible to work out the shift being used for various FSK signals.

In fact the resolution is almost good enough to be used as a QRSS (very slow morse) visual skimmer.

Here's a screen grab from part of the weak signal data modes section of the 10MHz amateur band.

Regards,

Martin.

[image]

jks-prv, 2017-01-17:
The v1.44 release adds z14 to the previous z12-13.

z15 worked, but the large decimation causes undesirable effects. The time resolution is poor because of the large overlapped sampling. This makes the waterfall look "streaky" in the vertical dimension. The delay in refilling the decimation pipeline when switching waterfall frequency is several seconds which is annoying. Both of these problems are due to the "acquisition time problem" discussed in the Kiwi design paper.

I added a couple of user interface hacks to try out. The zoom "max in" button now toggles between an intermediate, but useful, value (z10) and max-in (z14). I judged this better than trying to squeeze in another zoom button requiring a different icon. The button turns orange for z10-13 and green at the maximum z14. Similarly the zoom "max out" button turns green at z0.

If you hold the shift-key while pressing the zoom in/out icons (magnifying glasses) the passband will increase/decrease by 80% of its current size around the passband center frequency. This is much easier than trying to drag the passband edges when you just want to "narrow things up" a little bit.

jks-prv, 2017-01-17:
BTW, thanks for pointing out BBC R4 sending data. I didn't realize it did. I knew about TDF/ex-RFI on 162 kHz but always assumed it was the only one.

G8JNJ, 2017-01-17:
Hi John,

Both the new enhancements work very well.

Just a thought - I wonder if it's worth including a similar intermediate level of Max zoom so that you can quickly jump a few levels to see a whole band (perhaps zoom 5) without going to full zoom immediately ?

I think it would save a lot of 'up and down' zooming using just the plus and minus buttons.

Regards,

Martin

closed 2017-03-04 by jks-prv