jks-prv/Beagle_SDR_GPS #73: WSPR hangs after a short time in 1.28

slepp, 2016-12-11:
Since the 1.28 update, WSPR manages to run at most about 2 minutes before the extension stops working. JS console doesn't seem to indicate anything and the extension websocket is open, but no frames are showing.

jks-prv, 2016-12-11:
Will be fixed in v1.29 later today.

closed 2016-12-11 by jks-prv

slepp, 2016-12-11:
Excellent, thank you.

And mostly unrelated, I also plan to start work on some FPGA->PRU->WSPR decoding chain, since it seems there may be enough DSP resources for the DDC/filter chain needed there to allow for an always-on WSPR decoder.

jks-prv, 2016-12-11:
That sounds like fun. Note that there are already tools and support code in the distribution for server (C-code) to PRU manipulation (TI's PRU assembler, downloading, shared-memory communication, etc.) See the pru/ sub-directory. It is not used presently for anything. It came from my hp 5370 project where the PRU is used for real-time control of the 5370 instrument bus to avoid problems of missing bus events when Linux gets in the way.

slepp, 2016-12-11:
I saw the main parts of the compiler were there, and I read through the design documentation. I haven't made it as far as seeing which GPIO go to PRU0, but figured it'd be easy enough to just bang out the 375 sps it needs directly without FPGA buffering. If there's room for multiple DDC multipliers, sending for all the main WSPR channels could also be accomplished.

Just need to get Vivado working to start work with some of the Verilog ideas/figuring out how things interconnect. :)

jks-prv, 2016-12-12:
Yes, getting Vivado downloaded and running is always a headache. I have an open request to document how to run Vivado to build the Kiwi firmware. It hasn't come anywhere near the top of the priority list yet. I also checked the web for advice about exporting a Vivado project to a small number of files for use in a distribution. Incredibly, this is apparently not possible without some custom script hacks people have come up with. So this would be another big time sink to get working and verified.

One relatively easy thing I could do that is necessary for experienced Vivado users is documenting the settings of all the generated IP that the project uses.