jks-prv/Beagle_SDR_GPS #74: Ability to edit markers broken in version v1.28

G8JNJ, 2016-12-11:
Hi John,

I tried to edit some markers this evening, but It now asks for an Admin password even when logged in locally. 

I have set a password and tried "Admin auto-login from local net even if password set?" set to both "yes" and "no" with and without a password and it won't let me in.

The "User auto-login from local net even if password set?" seems to work correctly. Both with and without a password.

Regards,

Martin

jks-prv, 2016-12-12:
Will be fixed in v1.30

closed 2016-12-12 by jks-prv

G8JNJ, 2016-12-12:
Hi John,

I can now edit markers, but when I click Modify, Add or Delete to commit the Browser locks up.

I have to start a new session to be able to use the SDR.

Regards,

Martin

jks-prv, 2016-12-12:
Okay, another fix for v1.30
Given the fix this should have always been a problem. I don't quite understand why it wasn't. Something in changing the underlying protocol from AJAX to web sockets as triggered it. And I don't have time to look into it.