jks-prv/Beagle_SDR_GPS #76: Bug - Waterfall background level changes when editing tags with /?fn=2 switch using V1.31

G8JNJ, 2016-12-14:
Hi John,

This is a strange bug. 

When the SDR is running with the /?fn=2 switch is added to the URL and when I try to edit a tag, the waterfall background level changes and goes darker as soon as I add the first text character.

The FS Min slider setting doesn't change in value but if I move the slider even slightly, the waterfall background returns to normal.

Regards,

Martin

jks-prv, 2016-12-14:
"/?fn=2" ? Do you mean "/?f=2" ?
I fixed this identical bug for another set of circumstances, but I need to know exactly how to reproduce it for your case. I tried both fn=2 and f=2, edited the "ident" field of an existing tag and there was no problem. Thanks.

G8JNJ, 2016-12-14:
Hi John,

Sorry I messed up and got the switch syntax wrong I think it should have been /?sqrt=2.

Not sure how I managed to get this mixed up, but I compounded the problem by setting up a bookmark with the wrong Fn command.

I think you can close this one.

Regards,

Martin

jks-prv, 2016-12-14:
Well, let me know if you think this is a problem in v1.31
It was definitely a bug in v1.28, but got fixed beginning in v1.29

G8JNJ, 2016-12-22:
Hi John,

This problem has returned in V1.33. 

The waterfall background level changes as soon as the Tag edit box is opened. 

This happens even if I'm not using the /?sqrt=2 switch, but it's more noticeable if the function is being used. 

Regards,

Martin

jks-prv, 2016-12-22:
Okay, I finally see what's going on here after having trouble replicating the problem.
It will be fixed in the next release.

closed 2016-12-22 by jks-prv