jks-prv/Beagle_SDR_GPS #78: DX List dont save in AM mode

Orloff2, 2016-12-20:
Hi
I add a radio in AM and when i click on it, it is become in AMN mode.
I shit click, modify to AM, click modify, reclick on it, it become AMN
The only solution is to modify dx.json manualy, change to AM et remember to not modify after via browser.
as i see, its only with AM mode, other (lsb etc) work.

jks-prv, 2016-12-20:
Thanks. This will be fixed in the next release.
The menu API was changed and I missed this one.

G8JNJ, 2016-12-20:
Hi John,

The 'Active' and 'Interference' (or whatever is bottom of the categories on the list) don't seem to save either.

Maybe someone else could confirm too ?

Regards,

Martin

jks-prv, 2016-12-20:
Yes, same issue. Should also get fixed in the next release.

closed 2016-12-22 by jks-prv