jks-prv/Beagle_SDR_GPS #84: Tag notes field don't display when tag hovered over (Chrome browser / Win7)

G8JNJ, 2017-01-07:
Hi John,

This is another odd one.

I was editing tags last night using Chrome on Windows 7. When I hovered over each tag the notes field displayed OK.

This morning, although the 'Pointing finger' pops up, the notes do not display.

Everything works fine using Firefox.

I also notice that the notes don't pop up when using IOS on a tablet regardless of browser.

Regards,

Martin

jks-prv, 2017-01-07:
These "tooltips" in browsers have always been problematic. There is the old way of doing it, simply adding a "title" attribute to an HTML element. The browser is then supposed to handle all the details of showing the tooltip correctly on an element hover. But it doesn't always seem to be reliable.

Then there is a much more complicated way of implementing tooltips using CSS. That's what's done for the spectrum display dBm readout and passband parameters. Maybe I need to do this for the tag notes too.

I'm not sure but I don't think mobile browsers have any concept of a title attribute tooltip because there is no such thing as a cursor position -- only touch events. On my iPad CSS tooltip are completely interfered with by Safaris magnifying glass thing.

G8JNJ, 2017-01-08:
Hi John,

OK all understood.

It's very odd that it works sometimes but not others. 

I've done all the usual stuff like clearing the cache etc. but nothing obvious seems to fix it (or bring on the problem when it's working).

I'ts a bit annoying as I've just started tidying up and reducing the size of the text in my tags by adding the additional info to the notes field, but now it's being temperamental about displaying them :-( 

I'll just use Firefox for the time being.

Regards,

Martin

closed 2017-03-04 by jks-prv