jks-prv/Beagle_SDR_GPS #85: Use X-Forwarded-For header for connecing client IP

backburn, 2017-01-08:
When using a proxy like nginx to proxy port 80 to 8073, the application uses the ip of the proxy. In this case: 127.0.0.1

The application should use the X-Forwarded-For header for determining the client IP address when a local ip is seen. See https://en.wikipedia.org/wiki/X-Forwarded-For

Nginx config used to recreate issue (/etc/nginx/sites-enabled/default):
```
server {
    listen 80 default_server;
    listen [::]:80 default_server;
    server-name _;
    location / {
        proxy_pass http://127.0.0.1:8073;
        include /etc/nginx/proxy_params;
    }
}
```

OH1KK, 2017-01-10:
If you just port forward from 80 to 8073, you can also do that without nginx. Then you do not need X-Forwarded-For.

Type on KiwiSDR

sudo iptables -t nat -A PREROUTING -i eth0 -p tcp --dport 80 -j REDIRECT --to-port 8073

jks-prv, 2017-01-10:
Sorry, I'm a little confused about this issue. Is it only applicable for connections inside your firewall and not outside for the general Internet? Are you really just trying to eliminate the need to type the 8073 port number in the URL? If so, you can simply configure the Kiwi server to use port 80 instead. See the port field on the network tab of the admin page.

Now this is obviously a problem if your Kiwi is public but you're already mapping port 80 to some other web server on your network via a NAT rule on your firewall such that your public IP port 80 can't be mapped to the Kiwi.

backburn, 2017-02-05:
Nginx provides reverse-proxy capability. Benefits of a reverse proxy include:
- Protecting the IP address of the host by using nginx as a reverse proxy on an offsite server.
- Using paths like kiwisdr.k7biz.com/kiwi1, kiwisdr.k7biz.com/kiwi2, etc.. for multiple kiwis. 
- Load balancing requests to multiple devices.
- Rerouting ports (8073->80).

If the kiwi depends on the IP address of a user, it should support X-Forwarded-For headers. Otherwise commonly used security techniques like using a reverse proxy will cause issues.

benphelps, 2017-02-05:
A more realistic nginx configuration that exposes the issue at hand would be

```nginx
server {
    listen 80 default_server;
    listen [::]:80 default_server;
    server-name _;
    
    root /sites/example.com/public;

    location ~ [^/]\.php(/|$) {
        fastcgi_split_path_info ^(.+?\.php)(/.*)$;
        if (!-f $document_root$fastcgi_script_name) {
            return 404;
        }
        fastcgi_param HTTP_PROXY "";
        fastcgi_pass 127.0.0.1:9000;
        fastcgi_index index.php;
        include /etc/nginx/fastcgi_params;
    }

    location /kiwi {
        proxy_pass http://127.0.0.1:8073;
        include /etc/nginx/proxy_params;
    }
}
```

Where nginx is running as the web server (consuming port 80) and will proxy requests made to `/kiwi`.

benphelps, 2017-02-05:
Perhaps this is an upstream issue with the mongoose library, I've searched their documentation and source, finding nothing to handle proxied connections.

jks-prv, 2017-07-29:
Okay, massive apologies to @backburn for my poor understanding of this issue. This is actually very useful for the following case: We now have a pull request (https://github.com/jks-prv/Beagle_SDR_GPS/pull/107) from a guy in China who has successfully been running a Kiwi using the "frp" (https://github.com/fatedier/frp) reverse proxy because DDNS and opening incoming NAT ports is not an option for him. The proxy is hosted on a third-party machine that provides a public ip (i.e. not on his restrictive home network).

In his case he parses the X-Real-IP header sent by frp to set the client ip used by the Kiwi server. From what I've read it seems X-Forwarded-For may also sent by proxies and cacheing servers.

jks-prv, 2017-08-03:
There are a bunch of knowledgeable people on here so please indulge me a question.

One of the nice things about frp is that is has a proxy server _and_ client. The client can contact the server to get the connection started and this allows punching through an ISP-controlled firewall that disallows initial incoming connections. I understand the same can be achieved by using an nginx reverse proxy with an ssh (client) / sshd (server) tunnel. But this has the problem of the extra overhead of encrypting the Kiwi stream traffic over the tunnel which is not necessary. There are going to be enough problems with the extra overhead of a hop through a VPS.

Does anyone know a better solution to this? There doesn't seem to be an nginx module that allows it to also run on the client and act like a low-overhead tunnel back to another nginx running the reverse proxy on the VPS.

If we were to consider running a proxy service for Kiwi owners then nginx obviously has a better track record than frp which is relatively new. I've just started to look at the UDP-based Tinc VPN which might be a solution (and has been around forever).

benphelps, 2017-08-04:
So for our usage we would have a public facing sdr running on our home internet on an open port, nginx is running on a hosted VPS, the VPS connects to the sdr.  The connection comes with a `X-Forwarded-For` header, this header can allow the sdr to rate limit the connections as if they were made directly.  Otherwise they all appear as if they came from the hosted VPS.  No SSH is needed for us.

jks-prv, 2017-08-04:
That use case I understand. The tricky one is when the ISP has blocked any initial incoming connections (i.e. the first TCP SYN has to come from the client net inside the firewall). When nginx on the VPS in this case attempts to contact the kiwi server it will never get a response because there was nothing on the client side to first initiate the connection. frp works because the client part of frp (frpc) first contacts the frp server on the VPS (frps) and keeps a control channel open and polls it for new requests from users connecting to the VPS wanting to get to the Kiwi.

lazywalker, 2017-08-08:
[ngrok](https://ngrok.com) would do the same,  this one seems to have more users than frp, but if i have to choose i'll pick frp, for it's simply code and easy to modified to fix our needs. I have ran frpc-to-frps for about one month, with tmux and a simple startup shell script so they can auto-run with system boot. The services are quite stable without crash or any unexpected quit, except the connection lost between my router and the VPS sometimes, i believe that was caused by my ISP policy, because when connection lost i can't even ssh to my VPS. Well, I will use my secondly network service(provided by other ISP) to run the test for another month, we will see:)

jks-prv, 2017-08-08:
I had understood you had to license (and pay) to run the ngrok server on your own VPS. They don't quote pricing on their website which pretty much means it's unaffordable for our purposes I would think.

jks-prv, 2017-10-22:
The X-{Real-IP, Forwarded-For} code I added to support the new proxy.kiwisdr.com service seems to be working okay. So I'm going to close this issue unless someone thinks more work needs to be done.

closed 2017-10-22 by jks-prv