jks-prv/Beagle_SDR_GPS #87: SDR freezes when any of the Contact's links are used

G8JNJ, 2017-02-27:
Hi John,

I've just noticed that the SDR freezes when any of the Contact's links are used.

An F5 or refresh is required to restore operation.

Regards,

Martin - G8JNJ

jks-prv, 2017-02-28:
This is an issue with how Javascript interacts with "mailto:" URLs. I've tried and as far as I'm aware it cannot be fixed.

G8JNJ, 2017-02-28:
Hi John,

Ah OK understood.

Not a major problem but worth knowing.

Suggestion - I wonder if cursors or some other marker could be placed on the frequency scale in order to indicate where the other three receivers are tuned to when in use ? That would make it easy to click on them in order to see (and hear) what is being listened to ?

Regards,

Martin - G8JNJ

closed 2017-03-04 by jks-prv