jks-prv/Beagle_SDR_GPS #89: Feature request: query number of online users

OH1KK, 2017-03-04:
Hello.

With antenna switch extension scripting It would be handy to get information from KiwiSDR about currently connected users. With this information it could be possible to make cron script that resets default antenna configuration when no users are online. Or it would be possible to forbid antenna switching if there are more than one users online.

For shell script, info should be in easily querable and parseable format. For example.

`wget http://kiwisdr.com:8073/?q=userscount`

or
`/root/Beagle_SDR_GPS/kiwi-cli userscount`

Reply could be simpy as

`2 users connected.`

jks-prv, 2017-03-04:
The status URL needed by sdr.hu has this information now.
For example
[kiwisdr.jks.com:8073/status
](kiwisdr.jks.com:8073/status
)
produces:
```
status=active
name=0-30 MHz SDR, ZL/KF6VO | Tauranga, New Zealand
sdr_hw=KiwiSDR v1.58
op_email=jks@kiwisdr.com
bands=0-30000000
users=0
users_max=4
avatar_ctime=1488485628
gps=(-37.631016, 176.172019)
asl=30
loc=Tauranga, New Zealand
sw_version=KiwiSDR_v1.58
antenna=Mini-Whip
```

OH1KK, 2017-03-05:
Thank you. This is what I was looking for.

closed 2017-03-05 by jks-prv